sair-lab/AirSLAM #43: error: no match for ‘operator=’ (operand types are ‘std::shared_ptr<cv::ximgproc::FastLineDetector>’ and ‘cv::Ptr<cv::ximgproc::FastLineDetector>’)

ZJJ2022415, 2023-01-04:
请问catkin_make:时报错：该怎么解决？

AirVO_ws/src/AirVO-master/src/line_processor.cc:465:118: error: no match for ‘operator=’ (operand types are ‘std::shared_ptr<cv::ximgproc::FastLineDetector>’ and ‘cv::Ptr<cv::ximgproc::FastLineDetector>’)
    line_detector_config.canny_th1, line_detector_config.canny_th2, line_detector_config.canny_aperture_size, false);

xukuanHIT, 2023-01-04:
@ZJJ2022415 Hi, please tell me your test environments.

ZJJ2022415, 2023-01-06:
> @ZJJ2022415 Hi, please tell me your test environments.

cuda11.2 cudnn 8.6 eigen3.4 opencv3.4.11

xukuanHIT, 2023-01-07:
@ZJJ2022415 Hi, you have a different opencv version from us. It seems that the FastLineDetector API has changed, so you can try to solve it in 3 ways. 
1. Upgrade your opencv from 3.x to 4.x. 
2. Change the related code in AirVO according to the API of opencv 3.x. 
3. Use the docker recommended in the readme.

ZJJ2022415, 2023-01-13:
> 

After I successfully compiled in dokcer, why didn't I generate"(find air_vo)/debug/traj.txt" under the defined path

[image]

[image]

yuefanhao, 2023-01-14:
@ZJJ2022415 Hi, you can set the log level to **kERROR** [here](https://github.com/xukuanHIT/AirVO/blob/67d742ede8ad840674db102ed9acf1975def67cb/src/super_point.cpp#L16) and rebuild the project, which will show more detailed information about this error in the process. Thank you.