sair-lab/AirSLAM #45: How To convert SuperGlue model Correctly？

MirrorYu, 2023-01-05:
Great work! Thanks a lot! 
some details of the project I need your help!
I follow the step by convert2onnx subdirectory:
step 1: convert torch model to onnx format:
>> python convert_superglue_to_onnx.py
step 2: onnx simplyfy the onnx model:
>> python -m onnxsim ../output/superglue_indoor.onnx ../output/superglue_indoor_sim.onnx --dynamic-input-shape --input-shape keypoints_0:1,512,2 scores_0:1,512 descriptors_0:1,256,512 keypoints_1:1,512,2 scores_1:1,512 descriptors_1:1,256,512
step 3: convert int64 to int 32:
>> python convert_int32.py

now, I got three models:
[image]

when i run the code，an error will be occured in the methold "build" of super_glue:
[image]

but when i use your provided "superglue_outdoor_sim_int32.onnx" in output directory，everything works well.

SO,can you provide the process of  the model converted?

xukuanHIT, 2023-01-05:
@MirrorYu Hi, thank you for your interest. @yuefanhao will answer the question after finishing an important exam On this Saturday.

MirrorYu, 2023-01-05:
OK，Thanks a lot！

yuefanhao, 2023-01-09:
@MirrorYu Hi, did you use the [superglue official model](https://github.com/magicleap/SuperGluePretrainedNetwork) for the conversion? the model conversion process seems to be fine, can you provide more details about this error? You can set the log level at here [super_glue.cpp#L17](https://github.com/xukuanHIT/AirVO/blob/67d742ede8ad840674db102ed9acf1975def67cb/src/super_glue.cpp#L17) that you can see more details of the running information.

yuefanhao, 2023-01-09:
> @MirrorYu Hi, did you use the [superglue official model](https://github.com/magicleap/SuperGluePretrainedNetwork) for the conversion? the model conversion process seems to be fine, can you provide [...]
@MirrorYu Or you can try just doing the first conversion step and see if it works.

MirrorYu, 2023-01-09:
Thanks for your reply! I have checked:
1.model comes from official [website](https://github.com/magicleap/SuperGluePretrainedNetwork)；
2.the log is:
[image]
3.I want to check the process step:
（1）torch -> onnx;
（2）onnx -> onnxsim;
（3）int64 -> int32;
or
 （1）torch -> onnx;
（2）int64 -> int32;
（3）onnx -> onnxsim;
4.The model i converted is:
https://github.com/MirrorYu/Superglue_model
Can you help me to check it? Thanks a lot! Hope for your reply~

MirrorYu, 2023-01-09:
my system  is win10, conda information is:
（1） pytorch 
[image]
（2）onnx
[image]

yuefanhao, 2023-01-09:
@MirrorYu  Hi, the process step should be 
（1）torch -> onnx;
（2）onnx -> onnxsim;
（3）int64 -> int32;
I'll check your model and try to reproduce your problem, and I'll reply you as soon as I finished it.

MirrorYu, 2023-01-09:
OK，Hope for your reply！

yuefanhao, 2023-01-10:
@MirrorYu Hi, I used the converted model you provided to try to reproduce your problem, but it works well on my computer. There are some warnings in the running process, but they will not affect the final result. I ran it in the Docker environment we provided, using the OIVIO dataset. Here are two ways to verify your problem:

First, I have noticed that the version of some python libraries in our Docker environment may be different from yours, you can try to run it in our Docker. And some of our docker Python library information is as follows:
cuda==11.6
torch==1.9.1
onnx==1.11.0
onnx-simplifier==0.3.10
onnxoptimizer==0.2.7
onnxruntime==1.11.1
tensorrt==8.4.1.5

Second, it should be noted that the name of the model is configured correctly. The name of the model in our default [configuration file](https://github.com/xukuanHIT/AirVO/blob/67d742ede8ad840674db102ed9acf1975def67cb/configs/configs_oivio.yaml#L26) is _superglue_indoor_sim_int32.onnx_. I noticed that yours is _superglue_outdoor_sim_int32.onnx_, make sure you have modified it.

If it still can't be solved, you can take screenshots of the log of errors during the running process to me. Thank you.

MirrorYu, 2023-01-10:
OK，Got it， I'll try it later. Thanks again！

MirrorYu, 2023-01-11:
> @MirrorYu Hi, I used the converted model you provided to try to reproduce your problem, but it works well on my computer. There are some warnings in the running process, but they will not affect the [...]

That's Work!  I install Python library as you mentioned, The converted model works well! Thank you for your patient reply Again!