sair-lab/AirSLAM #52: Error in SuperPoint building

ZJJ2022415, 2023-02-01:
Hello, why are the following errors when I run launch after I compile successfully?

Error in SuperPoint building

MapBuilder::MapBuilder(Configs& configs): _init(false), _track_id(0), _line_track_id(0), 
    _to_update_local_map(false), _configs(configs){
  _camera = std::shared_ptr<Camera>(new Camera(configs.camera_config_path));
  _superpoint = std::shared_ptr<SuperPoint>(new SuperPoint(configs.superpoint_config));
  if (!_superpoint->build()){
    std::cout << "Error in SuperPoint building" << std::endl;
    exit(0);
  }

ZJJ2022415, 2023-02-01:
Excuse me, does my use of Tensorrt 8.4.1.5 affect compilation?

yuefanhao, 2023-02-03:
@ZJJ2022415 Hi，are you running in the docker environment we provide?

ZJJ2022415, 2023-02-04:
> @ZJJ2022415 Hi, are you running in the docker environment we provide?

I compiled successfully on my computer. thank.

closed 2023-02-07 by yuefanhao