sair-lab/AirSLAM #53: about runing enviroment

ynma-hanvo, 2023-02-15:
thanks for your great efforts.  will it run on  environment of   cuda 102  and ROS melodic?

xukuanHIT, 2023-02-15:
@ynma-hanvo Hi, thanks for your interest. We have not tested on the environments you mentioned. ROS melodic should be no problem because we don't use too many functions of ROS. But I'm not sure whether a lower cuda version is okay. As for the TensorRT, it's better to use v8.4.1.5.

ynma-hanvo, 2023-02-16:
> @ynma-hanvo Hi, thanks for your interest. We have not tested on the environments you mentioned. ROS melodic should be no problem because we don't use too many functions of ROS. But I'm not sure whether a lower cuda version is okay. As for the TensorRT, it's better to use v8.4.1.5.

thanks for your reply.  I will upgrade my workstation to cuda11.6 and test.

closed 2023-02-16 by ynma-hanvo