sair-lab/AirSLAM #57: docker: Error response from daemon: failed to create shim task

ZhiYiXinNan, 2023-03-03:
[image]
How can I debug these issues?

xukuanHIT, 2023-03-03:
@ZhiYiXinNan Hi, do you have installed the Nvidia driver and Nvidia docker?

ZhiYiXinNan, 2023-03-03:
I have installed the docker

ZhiYiXinNan, 2023-03-03:
[image]
I have installed nvidia driver ,but SMI can't communicate with the driver

ZhiYiXinNan, 2023-03-03:
my use is vmware workstation

xukuanHIT, 2023-03-04:
@ZhiYiXinNan Hi, docker and Nvidia-docker (https://github.com/NVIDIA/nvidia-docker) are different. I have not used VMware. It seems that the Nvidia driver is not available in your environment.

ZhiYiXinNan, 2023-03-05:
I am preparing change my enviroment,so,if I choose to use the docker you provite,should I need to install the dependencies like ROS ,OPenCV4.2.0,CUDA...?Or I should install these dependencies first then install docker and Nvidia-docker

ZhiYiXinNan, 2023-03-05:
I am preparing change my enviroment,so,if I choose to use the docker enviroment you provide,should I need to install the dependencies like ROS ,OPenCV4.2.0,CUDA 11.6...?Or I just need to run the two docker commands like this.
[image]

xukuanHIT, 2023-03-06:
@ZhiYiXinNan Hi, these dependencies have been integrated into the docker environment.

ZhiYiXinNan, 2023-03-06:
Appreciate!One more quesetion,if I want to change the code to do my research and test,will the docker enviroment also need to be changed? Or I just need to change the codes ,and run the two commands,then the codes I changed will run as I wish
[image]

ZhiYiXinNan, 2023-03-06:
Appreciate!One more quesetion,if I want to change the code to do my research and test,will the docker enviroment also need to be changed? Or I just need to change the codes ,and run the two commands,then the codes I changed will run as I wish

ZhiYiXinNan, 2023-03-06:
Does the docker environment include the data.like ovio euroc .etc....?

xukuanHIT, 2023-03-07:
@ZhiYiXinNan Hi, code and data are not in the docker.

ZhiYiXinNan, 2023-03-08:
hello,where should I put the data in? I have install the docker environment ,but i don't know what path  should I put the data in.

ZhiYiXinNan, 2023-03-08:
I try to put the euroc dataest under /home/zzj like this
[image]

ZhiYiXinNan, 2023-03-08:
[image]
Do I need to  put the data into the docker environment?

ZhiYiXinNan, 2023-03-08:
Do I need to put the data into the docker environment?
[image]

xukuanHIT, 2023-03-09:
@ZhiYiXinNan Hi, I think these two links may be helpful for you.
https://docs.docker.com/storage/

ZhiYiXinNan, 2023-03-09:
you mean i should put the dataest V1_02_medium in /media/xukuan/somethings2/ubuntu_files/data/euroc?

ZhiYiXinNan, 2023-03-09:
你好，我实在搞不明白这个dataroot为什么一直告诉我"doesn't exist "不存在，无论我是自己手动修改路径 还是尝试将数据集拷贝到容器里都失败了，请问有什么好的解决方法吗？

ZhiYiXinNan, 2023-03-09:
[image]
It stuck,what can I do?

ZhiYiXinNan, 2023-03-09:
[image]
when I run "roslaunch air_vo euroc.launch",it stuck for long time ,after waiting it shows that
[image]

ZhiYiXinNan, 2023-03-09:
and shows Failed when extracting point features
[image]

ZhiYiXinNan, 2023-03-09:
My GPU is GeForce RTX3070

ZhiYiXinNan, 2023-03-09:
Hello, I have deal the dataroot problem ,but when I run "roslaunch air_vo euroc.launch",it stuck for long time ,after waiting it shows that
[image]
and stuck for long time and told me failed when extracting point features

[image]
I am anticipating your help and will be appreciate for it

xukuanHIT, 2023-03-09:
@ZhiYiXinNan Hi, the first picture shows the code has run successfully and the trajectory has been saved.
For the problem in the second picture, please take a look at [this link](https://github.com/xukuanHIT/AirVO/issues/43#issuecomment-1382732091) to show more debug information.
For the long time stuck, please refer to #54 .

ZhiYiXinNan, 2023-03-09:
what about  the problem in first picture?
[image]

ZhiYiXinNan, 2023-03-10:
Yes ,you are right ,the first time I run the code is too slow ,then will be faster,but I still can not deal with the error,could you please tell me what wrong with it,Is my environment not true?

ZhiYiXinNan, 2023-03-10:
I glanced over the link you gave to me ,I found I don't generate "(find air_vo)/debug" in AirVO,what cause it?
[image]

ZhiYiXinNan, 2023-03-10:
[image]
how can I debug this problem?Appreciate for your  help

ZhiYiXinNan, 2023-03-10:
Is this problem caused by the code?
[image]

xukuanHIT, 2023-03-11:
@ZhiYiXinNan It seems that the code has finished running and then exits. I think you can ignore it.

ZhiYiXinNan, 2023-03-11:
But how can I See the results?

xukuanHIT, 2023-03-11:
@ZhiYiXinNan 1. Configure related paths in launch files to save trajectory.
2. Configure related ros topic names in "config files" to visualize the results in rviz

ZhiYiXinNan, 2023-03-11:
Hello,you mean I can ignore "process has died "?

ZhiYiXinNan, 2023-03-11:
Hello, do you mean I can ignore the "process has died"on screen,and it will not influence rviz running,so I just need to run rviz then I can see the result ,all I can do is  check the configure related path?

ZhiYiXinNan, 2023-03-11:
I  am appriciated  for your help ,but please allow me ask one more question,what software do you use to compare different SLAM scheme,and does the "process has died"on screen will influence my following research(compare different SLAM scheme)? Appriciate!

xukuanHIT, 2023-03-11:
@ZhiYiXinNan 1. [evo](https://github.com/MichaelGrupp/evo)
2. No

ZhiYiXinNan, 2023-03-11:
Thank you ,I will try as you said .

ZhiYiXinNan, 2023-03-11:
> @ZhiYiXinNan It seems that the code has finished running and then exits. I think you can ignore it.
I think about your advice,how could you make sure the code has run successfully Under the premise of showing “process has died”on screen?

ZhiYiXinNan, 2023-03-13:
How can I visualize the results in rviz? When I run "rosrun rviz rviz" It shows 
No protocol specified
qt.qpa.xcb: could not connect to display :0
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.

Do you know how to deal with the problem?Appreciated

xukuanHIT, 2023-03-13:
@ZhiYiXinNan #44

closed 2023-03-15 by ZhiYiXinNan