sair-lab/AirSLAM #59: Error in SuperPoint building

18969068147, 2023-03-16:
`root@d09797bf0618:/workspace# roslaunch air_vo oivio.launch 
... logging to /root/.ros/log/3ab20a38-c3a5-11ed-94a1-0242ac110002/roslaunch-d09797bf0618-806.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://d09797bf0618:42449/

SUMMARY
========

PARAMETERS
 * /air_vo/camera_config_path: /workspace/catkin...
 * /air_vo/config_path: /workspace/catkin...
 * /air_vo/dataroot: /workspace/oivio/...
 * /air_vo/model_dir: /workspace/catkin...
 * /air_vo/saving_dir: /workspace/catkin...
 * /air_vo/traj_path: /workspace/catkin...
 * /rosdistro: noetic
 * /rosversion: 1.15.14

NODES
  /
    air_vo (air_vo/air_vo)

ROS_MASTER_URI=http://localhost:11311

process[air_vo-1]: started with pid [843]
config_file = /workspace/catkin_ws/src/AirVO/configs/configs_oivio.yaml
path = /workspace/oivio/TN_100_GV_01/husky0/cam0/data
[03/16/2023-10:51:34] [I] [TRT] [MemUsageChange] Init CUDA: CPU +572, GPU +0, now: CPU 608, GPU 980 (MiB)
[03/16/2023-10:51:35] [I] [TRT] [MemUsageChange] Init builder kernel library: CPU +1, GPU +0, now: CPU 628, GPU 980 (MiB)
[03/16/2023-10:51:35] [I] [TRT] ----------------------------------------------------------------
[03/16/2023-10:51:35] [I] [TRT] Input filename:   /workspace/catkin_ws/src/AirVO/output/superpoint_v1_sim_int32.onnx
[03/16/2023-10:51:35] [I] [TRT] ONNX IR version:  0.0.8
[03/16/2023-10:51:35] [I] [TRT] Opset version:    12
[03/16/2023-10:51:35] [I] [TRT] Producer name:    onnx-typecast
[03/16/2023-10:51:35] [I] [TRT] Producer version: 
[03/16/2023-10:51:35] [I] [TRT] Domain:           
[03/16/2023-10:51:35] [I] [TRT] Model version:    0
[03/16/2023-10:51:35] [I] [TRT] Doc string:       
[03/16/2023-10:51:35] [I] [TRT] ----------------------------------------------------------------
[03/16/2023-10:51:35] [W] [TRT] onnx2trt_utils.cpp:369: Your ONNX model has been generated with INT64 weights, while TensorRT does not natively support INT64. Attempting to cast down to INT32.
[03/16/2023-10:51:36] [I] [TRT] [MemUsageChange] Init cuBLAS/cuBLASLt: CPU +1340, GPU +378, now: CPU 1976, GPU 1358 (MiB)
[03/16/2023-10:51:36] [I] [TRT] [MemUsageChange] Init cuDNN: CPU +254, GPU +60, now: CPU 2230, GPU 1418 (MiB)
[03/16/2023-10:51:36] [W] [TRT] TensorRT was linked against cuDNN 8.4.1 but loaded cuDNN 8.4.0
[03/16/2023-10:51:36] [I] [TRT] Local timing cache in use. Profiling results in this builder pass will not be stored.
[03/16/2023-10:51:42] [E] [TRT] 1: [caskBuilderUtils.cpp::trtSmToCaskCCV::548] Error Code 1: Internal Error (Unsupported SM: 0x809)
[03/16/2023-10:51:42] [E] [TRT] 2: [builder.cpp::buildSerializedNetwork::636] Error Code 2: Internal Error (Assertion engine != nullptr failed. )
Error in SuperPoint building
[air_vo-1] process has finished cleanly
log file: /root/.ros/log/3ab20a38-c3a5-11ed-94a1-0242ac110002/air_vo-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
root@d09797bf0618:/workspace# 
`

18969068147, 2023-03-16:
I found that when  I using the different data sequence the log information can be different,  Another  error is that the point feature extraction is failed, is there any methods to find where the issue produced ?  the engine size probably the same 3MiB

yuefanhao, 2023-03-16:
@18969068147 Hi，You can try to delete the old engine file, rerun the code, and it will generate the new engine file automatically, and see if it works.

18969068147, 2023-03-17:
I delete all the files of /catkin_ws，and  it does works.  really thanks so much.

closed 2023-03-19 by yuefanhao