sair-lab/AirSLAM #63: Error in SuperPoint building

Pudit, 2023-06-29:
I tried to run uma_bumblebee_indoor.launch and euroc_ros.launch on docker as you recommend, but got the SuperPoint building error. 

root@arl-Thin-GF63-12VE:/workspace# roslaunch air_vo uma_bumblebee_indoor.launch 
... logging to /root/.ros/log/83057fb2-1671-11ee-bd38-adcfe6f4f8ef/roslaunch-arl-Thin-GF63-12VE-2796.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://arl-Thin-GF63-12VE:33725/

SUMMARY
========

PARAMETERS
 * /air_vo/camera_config_path: /workspace/src/Ai...
 * /air_vo/config_path: /workspace/src/Ai...
 * /air_vo/dataroot: /workspace/src/Ai...
 * /air_vo/model_dir: /workspace/src/Ai...
 * /air_vo/saving_dir: /workspace/src/Ai...
 * /air_vo/traj_path: /workspace/src/Ai...
 * /rosdistro: noetic
 * /rosversion: 1.15.14

NODES
  /
    air_vo (air_vo/air_vo)

ROS_MASTER_URI=http://localhost:11311

process[air_vo-1]: started with pid [2815]
config_file = /workspace/src/AirVO/configs/configs_uma_bumblebee_indoor.yaml
path = /workspace/src/AirVO/dataset/third-floor-csc2_2019-03-04-20-32-22_IllChange/cam0/data
Error in SuperPoint building
[air_vo-1] process has finished cleanly
log file: /root/.ros/log/83057fb2-1671-11ee-bd38-adcfe6f4f8ef/air_vo-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

xukuanHIT, 2023-06-30:
@Pudit Hi, could you change "kINTERNAL_ERROR" to "kINFO" in [this line](https://github.com/sair-lab/AirVO/blob/19569d180e4b4b4e2aaa8381916fd599e62b882c/src/super_point.cpp#L16C45-L16C60) to print more debugging information? You need to delete the old ".engine" files in "output" and re-generate them.

Pudit, 2023-06-30:
@xukuanHIT Thank you for your fast response! 
However, I cannot find any .engine files in output directory. 
How can I generate those files? 

This is the problem after following your advice without any those engine files:

> config_file = /workspace/src/AirVO/configs/configs_uma_bumblebee_indoor.yaml
path = /workspace/src/AirVO/dataset/third-floor-csc2_2019-03-04-20-32-22_IllChange/cam0/data
[06/30/2023-12:38:25] [I] [TRT] [MemUsageChange] Init CUDA: CPU +331, GPU +0, now: CPU 364, GPU 229 (MiB)
[06/30/2023-12:38:26] [I] [TRT] [MemUsageChange] Init builder kernel library: CPU +1, GPU +0, now: CPU 384, GPU 229 (MiB)
[06/30/2023-12:38:26] [I] [TRT] ----------------------------------------------------------------
[06/30/2023-12:38:26] [I] [TRT] Input filename:   /workspace/src/AirVO/output/superpoint_v1_sim_int32.onnx
[06/30/2023-12:38:26] [I] [TRT] ONNX IR version:  0.0.8
[06/30/2023-12:38:26] [I] [TRT] Opset version:    12
[06/30/2023-12:38:26] [I] [TRT] Producer name:    onnx-typecast
[06/30/2023-12:38:26] [I] [TRT] Producer version: 
[06/30/2023-12:38:26] [I] [TRT] Domain:           
[06/30/2023-12:38:26] [I] [TRT] Model version:    0
[06/30/2023-12:38:26] [I] [TRT] Doc string:       
[06/30/2023-12:38:26] [I] [TRT] ----------------------------------------------------------------
[06/30/2023-12:38:26] [W] [TRT] onnx2trt_utils.cpp:369: Your ONNX model has been generated with INT64 weights, while TensorRT does not natively support INT64. Attempting to cast down to INT32.
[06/30/2023-12:38:26] [I] [TRT] [MemUsageChange] Init cuBLAS/cuBLASLt: CPU +873, GPU +380, now: CPU 1265, GPU 609 (MiB)
[06/30/2023-12:38:26] [I] [TRT] [MemUsageChange] Init cuDNN: CPU +127, GPU +58, now: CPU 1393, GPU 667 (MiB)
[06/30/2023-12:38:26] [W] [TRT] TensorRT was linked against cuDNN 8.4.1 but loaded cuDNN 8.4.0
[06/30/2023-12:38:26] [I] [TRT] Local timing cache in use. Profiling results in this builder pass will not be stored.
[06/30/2023-12:38:35] [E] [TRT] 1: [caskBuilderUtils.cpp::trtSmToCaskCCV::548] Error Code 1: Internal Error (Unsupported SM: 0x809)
[06/30/2023-12:38:35] [E] [TRT] 2: [builder.cpp::buildSerializedNetwork::636] Error Code 2: Internal Error (Assertion engine != nullptr failed. )
Error in SuperPoint building

xukuanHIT, 2023-06-30:
@Pudit Could you tell me your GPU and driver version?

Pudit, 2023-06-30:
@xukuanHIT my gpu is NVIDIA GeForce RTX 4050 Laptop GPU, while 525.116.04 for driver version.

xukuanHIT, 2023-06-30:
@Pudit TensorRT 8.4 is not supported on Nvidia RTX 40 series. I have made another docker image for 40x GPU, you can delete the old image and try the new with "docker pull xukuanhit/air_slam:v3"

Pudit, 2023-06-30:
@xukuanHIT Thanks a lot, uma_bumblebee_indoor.launch and euroc.launch work pretty well. However, for euroc_ros.lauch has an issue about rviz, but I can public ros messages by adding --net=host.

closed 2023-06-30 by Pudit