sair-lab/AirSLAM #64: Compilation failure issues in Ubuntu18.04

zhongxiangrong, 2023-06-30:
My compilation environment is:
ubuntu18.04
g++ and gcc 7.5.0
cmake 3.24

The following error occurred when I executed the command "catkin_make"：

In file included from /usr/local/include/g2o/core/base_fixed_sized_edge.h:39:0,
                 from /usr/local/include/g2o/core/base_binary_edge.h:30,
                 from /usr/local/include/g2o/types/slam3d/types_slam3d.h:31,
                 from /home/uestc213/data/airvio_ws/src/AirVO/include/utils.h:25,
                 from /home/uestc213/data/airvio_ws/src/AirVO/src/camera.cc:4:
/usr/local/include/g2o/stuff/tuple_tools.h: In function ‘void g2o::tuple_apply_i(F&&, T&, int)’:
/usr/local/include/g2o/stuff/tuple_tools.h:45:35: error: ‘tuple_size_v’ is not a member of ‘std’
     std::make_index_sequence<std::tuple_size_v<std::decay_t<T>>>());
                                   ^~~~~~~~~~~~
/usr/local/include/g2o/stuff/tuple_tools.h:45:35: note: suggested alternative: ‘tuple_size’

So I made change to the CMakelist.txt:
`set(CMAKE_CXX_STANDARD 17)`

However ,  another error occurred:
error: ‘make_unique’ is not a member of ‘g2o’; did you mean ‘std::make_unique’?

xukuanHIT, 2023-06-30:
@zhongxiangrong Hi, it seems that some APIs of g2o have changed. Could you try to uninstall your g2o and install [this version](https://github.com/RainerKuemmerle/g2o/tree/20230223_git)

zhongxiangrong, 2023-06-30:
Another error:
/usr/local/include/g2o/core/base_variable_sized_edge.hpp:90:12: error: ‘class ceres::internal::FixedArray<double>’ has no member named ‘fill’
     add_vi.fill(0.);

It seems to be a problem with the version of Ceres being used.I am using Ceres1.14

xukuanHIT, 2023-06-30:
@zhongxiangrong Yeah, please try v2.0.0

zhongxiangrong, 2023-06-30:
install  ceres2.0.0  in PATH:  /usr/local/ceres2.0.0

Then , I changed cmakelist

It seems like the compilation has passed successfully. Thank you! I will try to run it later, and if I encounter any other issues, I will open a new issue.

closed 2023-06-30 by zhongxiangrong