sair-lab/AirSLAM #66: Windows, without ROS?

antithing, 2023-06-30:
Hi! Thank you for this code. Would it be possible to run it on Windows, without ROS?

xukuanHIT, 2023-06-30:
@antithing Hi, thanks for your interest. AirVO is unable to run on Windows without ROS now.

antithing, 2023-07-05:
Hi @xukuanHIT thanks! I have modified the code to compile on Windows, and I have it running on the euroc dataset.

However, i see this message:

`Failed when extracting point features !
`
about every five frames. 

It looks like this happens when inference fails in Superpoint, but why would this be happening?

Do you have any ideas here?

Thank you!

antithing, 2023-07-05:
Ah, this was an error with my dataset loading modifications, I have it running well now! .

closed 2023-07-05 by antithing