sair-lab/AirSLAM #67: roslaunch crashed

zhongxiangrong, 2023-07-03:
When I run the code "roslaunch air_vo euroc_ros.launch ", but I get an error：

setting /run_id to b3188950-19ab-11ee-8292-ac1f6ba0835e
process[rosout-1]: started with pid [37919]
started core service [/rosout]
process[air_vo_ros-2]: started with pid [37925]
process[rviz-3]: started with pid [37927]
config_file = /home/213/data/airvo_ws/src/AirVO/configs/configs_euroc.yaml
[ INFO] [1688393573.997696746]: rviz version 1.13.29
[ INFO] [1688393573.997757365]: compiled against Qt version 5.9.5
[ INFO] [1688393573.997776307]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1688393574.002567755]: Forcing OpenGl version 0.
[air_vo_ros-2] process has died [pid 37925, exit code -11, cmd /home/213/data/airvo_ws/devel/lib/air_vo/air_vo_ros __name:=air_vo_ros __log:=/home/213/.ros/log/b3188950-19ab-11ee-8292-ac1f6ba0835e/air_vo_ros-2.log].
log file: /home/213/.ros/log/b3188950-19ab-11ee-8292-ac1f6ba0835e/air_vo_ros-2*.log
[ INFO] [1688393574.461469073]: Stereo is NOT SUPPORTED
[ INFO] [1688393574.461561817]: OpenGL device: NVIDIA GeForce RTX 2080 Ti/PCIe/SSE2
[ INFO] [1688393574.461588794]: OpenGl version: 4.6 (GLSL 4.6).

I only modified the dataroot in the launch file.

In addition, can you provide the offline download of OIVIO？ Because the official website of OIVIO cannot be downloaded now.
thanks！！

zhongxiangrong, 2023-07-04:
I have solved it. Because the cv_bridge of ros melodic does not point to opencv4.2.

xukuanHIT, 2023-07-04:
Hi @zhongxiangrong , sorry for the late reply. I have uploaded OIVIO datasets to OneDrive. You can download using this [link](https://entuedu-my.sharepoint.com/:f:/g/personal/kuan_xu_staff_main_ntu_edu_sg/EtabKNOPBPBMmjJXbN4A8wMB0MTZmeZlnfAi7bskgfAr7A?e=JoIyDW).

xukuanHIT, 2023-07-08:
@zhongxiangrong Hi, we only use these sequences in the paper because they have the trajectory groundtruth.

HJMGARMIN, 2023-07-11:
@zhongxiangrong Hello, I met the same problem with ros melodic. How do you solve it? Thanks!

zhongxiangrong, 2023-07-13:
> @zhongxiangrong Hello, I met the same problem with ros melodic. How do you solve it? Thanks!

这是因为cv_bridge不兼容。你可以参考我的博客：https://blog.csdn.net/llllldm/article/details/126334841

closed 2023-07-24 by xukuanHIT