sair-lab/AirSLAM #70: UMA-VI Dataset

HJMGARMIN, 2023-07-11:


xukuanHIT, 2023-07-12:
@HJMGARMIN Hi, I only have a partial sequence. You can use this [link ](https://entuedu-my.sharepoint.com/:f:/g/personal/kuan_xu_staff_main_ntu_edu_sg/Emr16YVWbjVNhjaMt8qRGHABr2R_D_-WcHUvJDDePtNsGA?e=FOO2R5)to download the four sequences used in the paper.

HJMGARMIN, 2023-07-12:
Thanks for your reply. How did you download the four sequences? Do you have the link that you used?

xukuanHIT, 2023-07-12:
I previously requested the download link for the indoor illumination variation dataset from its author.

zhongxiangrong, 2023-07-13:
I have contacted the corresponding author of OIVIO. He stated that it may be due to changes in the server supplier. He will solve this problem as soon as possible.

> @xukuanHIT Do you have any link of OIVIO dataset? The original links are not available for public now (https://arpg.github.io/oivio/trajectories/)

xukuanHIT, 2023-07-13:
@quachconghoang Hi, I have uploaded some sequences to OneDrive. You can download using this [link](https://entuedu-my.sharepoint.com/:f:/g/personal/kuan_xu_staff_main_ntu_edu_sg/EtabKNOPBPBMmjJXbN4A8wMB0MTZmeZlnfAi7bskgfAr7A?e=Jn0S3B).

quachconghoang, 2023-07-14:
Many thanks @xukuanHIT, @zhongxiangrong

closed 2023-07-21 by xukuanHIT