sair-lab/AirSLAM #71: Is it possible to adapt AirVO with monocular cameras?

Pudit, 2023-07-12:
I've already tried feeding the same images from cam0 and cam1 and using the euro configuration without any modification. Fortunately, although stereo matching is one of the core of AirVO, it can localize a little bit. However, it's still stuck at its initial position as well as generating point clouds at wrong locations.

zhongxiangrong, 2023-07-13:
If the left and right images are the same image, Airvo cannot triangulate and therefore cannot locate. So currently only Stereo is supported.

xukuanHIT, 2023-07-13:
> If the left and right images are the same image, Airvo cannot triangulate and therefore cannot locate. So currently only Stereo is supported.

Yeah, that's correct.

Pudit, 2023-07-14:
I see. Do you guys have any plan to introduce some algorithms to make airvo work with monocular?

Pudit, 2023-07-19:
@xukuanHIT Thanks! I have one more question. When will the next version of AirVO (with loop closure, map reuse, and so on) be released?

xukuanHIT, 2023-07-19:
@Pudit A new stereo-VIO version will be released first by the end of this year. We plan to finish the full SLAM version within one year.

Pudit, 2023-07-19:
@xukuanHIT Sound great!! couldn't wait more! Just wondering, are there any potential possibilities to discuss research collaboration? I'm currently conducting research on Thermal SLAM.

xukuanHIT, 2023-07-21:
@Pudit Thanks for your interest. What form of collaboration are you referring to?

Pudit, 2023-07-22:
@xukuanHIT What kind of that I mentioned is research collaboration between your laboratory and mine. Please allow me to contact and provide you more detail. If you don't mind, I'll send it to your email shown on your paper.

xukuanHIT, 2023-07-23:
@Pudit OK, please provide me with more details, and I will be able to discuss it with my supervisor and other co-authors.

Pudit, 2023-07-23:
@xukuanHIT Thank you for this opportunity. I've already sent you an email.

closed 2023-07-23 by Pudit