sair-lab/AirSLAM #78: _keyframe_ids.size ==0 why?

haha911, 2023-07-22:
I want to ask a question!!
first,I use below command:
docker pull xukuanhit/air_slam:v1
docker run -it --env DISPLAY=$DISPLAY --volume /tmp/.X11-unix:/tmp/.X11-unix --privileged --runtime nvidia --gpus all --volume ${PWD}:/workspace --workdir /workspace --name air_slam xukuanhit/air_slam:v1 /bin/bash
I use 2080TI  gpu;
second . I use the command and all success
[image]
third ,I use command "roslaunch air_vo uma_bumblebee_indoor.launch "
but get the information:
[image]

[image]

I want to ask why "_keyframe_ids.size = 0"  ?? thank you!!

haha911, 2023-07-22:
my   uma_bumblebee_indoor.launch file as belows:
[image]

closed 2023-07-22 by haha911

2805651606, 2024-10-12:
> 我想问一个问题！！ 首先，我使用以下命令： docker pull xukuanhit/air_slam:v1 docker run -it --env DISPLAY=$DISPLAY --volume /tmp/.X11-unix:/tmp/.X11-unix --privileged --runtime nvidia --gpus all --volume ${PWD}:/workspace --workdir /workspace --name air_slam xukuanhit/air_slam:v1 /bin/bash 我使用 2080TI gpu； 其次，我使用该命令并且全部成功 第三，我使用命令“roslaunch air_vo uma_bumblebee_indoor.launch ” 但得到以下信息： ![图像](https://private-user-images.githubusercontent.com/29395182/255332896-fa8c3b22-dc6f-4b4c-9609-d4327077f753.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg3MDA0ODgsIm5iZiI6MTcyODcwMDE4OCwicGF0aCI6Ii8yOTM5NTE4Mi8yNTUzMzI4OTYtZmE4YzNiMjItZGM2Zi00YjRjLTk2MDktZDQzMjcwNzdmNzUzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDEyVDAyMjk0OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWEyNGNlNjI5Y2JlMDlkNDhkZjdjOTEwOGEwMTliODI3YzE5NjIwYTE0OWFjNTliMzZlZGZhYmQ4YzcwYWU2MGQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.YlOr4XF2eLJ0-LDy3lYJGCge6honyreCIAYBz6dSEGo) ![图像](https://private-user-images.githubusercontent.com/29395182/255333069-e3514871-6fe2-4432-9155-d818f69238a0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg3MDA0ODgsIm5iZiI6MTcyODcwMDE4OCwicGF0aCI6Ii8yOTM5NTE4Mi8yNTUzMzMwNjktZTM1MTQ4NzEtNmZlMi00NDMyLTkxNTUtZDgxOGY2OTIzOGEwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDEyVDAyMjk0OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTc4YzFkNjQxYzBmMDRiNjUxYzI0NzBlZDlhMGEzM2E3NWVkY2VkMjdhNGMyODE3Njg3ZTU2ZjAwZmU0MWY4MWImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.TNDSE9asZ07h12YfTRvLM2xE63y3Ab6BJGliPB9jIjo)
> 
> ![图像](https://private-user-images.githubusercontent.com/29395182/255333009-b33d2997-800f-4422-a428-32d6f2670b90.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg3MDA0ODgsIm5iZiI6MTcyODcwMDE4OCwicGF0aCI6Ii8yOTM5NTE4Mi8yNTUzMzMwMDktYjMzZDI5OTctODAwZi00NDIyLWE0MjgtMzJkNmYyNjcwYjkwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDEyVDAyMjk0OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTExODI4NzUzZjMyMmMyMDY2MzRmZjg3MWYyYjNjZDFkMzRkMDIzYzI1OTUzOGEwNGZlZDQyNWI3MDhiMTJkMDUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.ovg5CD31S84RHo_Ip9MJ3vnQrHbVXLEMQuB5B7aPqwY)
> 
> 我想问为什么“_keyframe_ids.size = 0”??谢谢！！

你好，请问你怎么解决的呢，我也是遇到了同样的问题