sair-lab/AirSLAM #79: roslaunch euroc.launch ;process has died

maxibooksiyi, 2023-08-18:
在 jetson Orin NX上，opencv 4.5.4   CUDA 11.4.315  TensorRT 8.5.2.2
部署编译都通过没有问题，数据集也下载了，在launch文件里面路径也改为对应的了，相同环境下运行过SuperPoint-SuperGlue-TensorRT没有问题
运行roslaunch air_vo euroc_ros.launch时 ,最后显示 process has died 
`nvidia@nvidia-desktop:~/AirVO_ws/devel$ roslaunch air_vo euroc.launch
... logging to /home/nvidia/.ros/log/7bea6782-3d8c-11ee-858e-788a8639a8d0/roslaunch-nvidia-desktop-25251.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nvidia-desktop:33595/

SUMMARY
========

PARAMETERS
 * /air_vo/camera_config_path: /home/nvidia/AirV...
 * /air_vo/config_path: /home/nvidia/AirV...
 * /air_vo/dataroot: /home/nvidia/MH_0...
 * /air_vo/model_dir: /home/nvidia/AirV...
 * /air_vo/saving_dir: /home/nvidia/AirV...
 * /air_vo/traj_path: /home/nvidia/AirV...
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    air_vo (air_vo/air_vo)

auto-starting new master
process[master]: started with pid [25259]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 7bea6782-3d8c-11ee-858e-788a8639a8d0
process[rosout-1]: started with pid [25269]
started core service [/rosout]
process[air_vo-2]: started with pid [25272]
config_file = /home/nvidia/AirVO_ws/src/AirVO/configs/configs_euroc.yaml
path = /home/nvidia/MH_01_easy/mav0/cam0/data
[air_vo-2] process has died [pid 25272, exit code -11, cmd /home/nvidia/AirVO_ws/devel/lib/air_vo/air_vo __name:=air_vo __log:=/home/nvidia/.ros/log/7bea6782-3d8c-11ee-858e-788a8639a8d0/air_vo-2.log].
log file: /home/nvidia/.ros/log/7bea6782-3d8c-11ee-858e-788a8639a8d0/air_vo-2*.log
`

maxibooksiyi, 2023-08-18:
[image]
[image]

xukuanHIT, 2023-08-18:
Log 给的信息有点少，可以debug到具体位置吗？看起来像是在[这个函数](https://github.com/sair-lab/AirVO/blob/8bc7fd571af1f89b039b8b99ec1690b619ae1f0e/src/dataset.cc#L8C4-L8C4)内部

maxibooksiyi, 2023-08-21:
> Log 给的信息有点少，可以debug到具体位置吗？看起来像是在[这个函数](https://github.com/sair-lab/AirVO/blob/8bc7fd571af1f89b039b8b99ec1690b619ae1f0e/src/dataset.cc#L8C4-L8C4)内部

下面是分别启动euroc.launch和euroc_ros.launch，gdb所打印的信息..
[image]
[image]

xukuanHIT, 2023-08-23:
可以检查下[这个函数](https://github.com/sair-lab/AirVO/blob/8bc7fd571af1f89b039b8b99ec1690b619ae1f0e/src/camera.cc#L53)的输入有没有问题

maxibooksiyi, 2023-08-27:
上面的报错我暂时通过注释掉camera.cc里的cv::initUndistortRectifyMap函数绕过了，然后就可以正常运行跑起来了，但是用euroc.launch跑euroc数据集跑完最后显示如下图所示，我看这里https://github.com/sair-lab/AirVO/issues/78别人也有遇到相同的情况
```
Save file to /home/nvidia/AirVO_ws/src/AirVO/debug/traj.txt
_keyframe_ids.size = 0
terminate called without an active exception
[air_vo-2] process has died [pid 135559, exit code -6, cmd /home/nvidia/AirVO_ws/devel/lib/air_vo/air_vo __name:=air_vo __log:=/home/nvidia/.ros/log/1db9f3f2-44c7-11ee-ac14-02426611b915/air_vo-2.log].
log file: /home/nvidia/.ros/log/1db9f3f2-44c7-11ee-ac14-02426611b915/air_vo-2*.log
```
[image]
gdb打印的信息如下
[image]

xukuanHIT, 2023-08-27:
红字是出现在序列跑完后的，不会影响最终结果。但最后keyfreame数量为0，有可能是最开始就跟丢了。initUndistortRectifyMap函数的问题最好能排查出来，比如你可以单独写一小段代码测试下在这个平台上的去畸变

maxibooksiyi, 2023-08-27:
> 红字是出现在序列跑完后的，不会影响最终结果。但最后keyfreame数量为0，有可能是最开始就跟丢了。initUndistortRectifyMap函数的问题最好能排查出来，比如你可以单独写一小段代码测试下在这个平台上的去畸变

好的好的，非常感谢，我进一步排查下initUndistortRectifyMap函数

xukuanHIT, 2023-09-15:
@maxibooksiyi Hi, 问题解决了吗？

closed 2023-09-15 by maxibooksiyi

2805651606, 2024-10-12:
@maxibooksiyi  請問這個問題你解決了？