sair-lab/AirSLAM #84: 我想请问一下，关于OIVIO数据集，您是怎么评估的，我没有找到评估相关的代码

Livesoso, 2023-10-28:


xukuanHIT, 2023-10-30:
用的[evo](https://github.com/MichaelGrupp/evo)

closed 2023-10-30 by Livesoso

xukuanHIT, 2023-10-31:
leica0里面是真值。AirVO保存的是TUM格式

Livesoso, 2023-10-31:
> leica0里面是真值。AirVO保存的是TUM格式

data.csv 里面是xyz，怎么转化为tum格式啊，辛苦您了（这个问题可能比较初级）

xukuanHIT, 2023-10-31:
可以写一个脚本，把旋转部分补成 0，0，0，1

Livesoso, 2023-10-31:
原来是这样
辛苦您了

BillyB233, 2024-10-03:
前辈们好，我发现OIVIO官方的下载链接404了，请问前辈们是否有其他下载途径呢？

2805651606, 2024-10-10:
> 前辈们好，我发现OIVIO官方的下载链接404了，请问前辈们是否有其他下载途径呢？

你好，请问你现在有数据集了？

Livesoso, 2024-10-11:
我之前找到了，具体从哪里下的我忘记了
我U盘里有
下班可以发你

BillyB233, 2024-10-11:
好的好的，十分感谢，这是我的邮箱czx0317@foxmail.com，祝你工作顺利

2805651606, 2024-10-12:
非常感谢，这是我的邮箱2805651606@qq.com,谢谢

Savior1s, 2024-11-10:
> 我之前找到了，具体从哪里下的我忘记了 我U盘里有 下班可以发你

麻烦您可以给我也发一份OIVIO数据集嘛，387725213@qq.com 感谢您们

nishibenbi, 2024-11-11:
> > 我之前找到了，具体从哪里下的我忘记了 我U盘里有 下班可以发你
> 
> 麻烦您可以给我也发一份OIVIO数据集嘛，[387725213@qq.com](mailto:387725213@qq.com) 感谢您们

请问有收到数据集吗，可以发我一份吗，非常感谢

sukaixiong, 2024-11-18:
> > > 我之前找到了，具体从哪里下的我忘记了 我U盘里有 下班可以发你
> > 
> > 
> > 麻烦您可以给我也发一份OIVIO数据集嘛，[387725213@qq.com](mailto:387725213@qq.com) 感谢您们
> 
> 请问有收到数据集吗，可以发我一份吗，非常感谢

请问您收到了吗，麻烦给我也发一份，十分感谢 2656419064@qq.com

laplace-pym, 2025-01-21:
> 我之前找到了，具体从哪里下的我忘记了 我U盘里有 下班可以发你

您好，请问可以发一下OIVIO 数据集吗，这是我的邮箱1721179144@qq.com，十分感谢！！！！

Livesoso, 2025-01-22:
https://github.com/sair-lab/AirSLAM/issues/175#issuecomment-2516008579

Livesoso, 2025-01-22:
https://entuedu-my.sharepoint.com/:f:/g/personal/kuan_xu_staff_main_ntu_edu_sg/EtabKNOPBPBMmjJXbN4A8wMB0MTZmeZlnfAi7bskgfAr7A?e=k62L8q