sair-lab/AirSLAM #90: camera_config

t676290894, 2023-12-05:
"Thank you for your work. As a beginner, I have learned a lot from it. I am not very familiar with some parameters in your camera intrinsic file. I understand that LEFT.D and LEFT.K are obtained from the dataset. Could you please explain how LEFT.R and LEFT.P are obtained?"
[image]

xukuanHIT, 2023-12-06:
@t676290894 Hi, they are outputs of [stereoRectify](https://docs.opencv.org/4.2.0/d9/d0c/group__calib3d.html#ga617b1685d4059c6040827800e72ad2b6).

t676290894, 2023-12-06:
Thanks sincerely for your help

xukuanHIT, 2023-12-09:
@t676290894 It's the output of [evo_ape](https://github.com/MichaelGrupp/evo/wiki/Metrics)

t676290894, 2023-12-09:
Thank you sincerely for your answer! I will close this issue.

closed 2023-12-09 by t676290894