sair-lab/AirSLAM #94: About the parameters of the algorithm？

zhongxiangrong, 2024-01-15:
What are the meanings represented by each parameter in the configs_euroc.yaml file?
I have found that in some scenarios, although line features increase robustness, they also improve translation errors. I guess it's the linear merge parameter that affects the results. I am trying to modify the parameters in the hope of achieving better results. But most of the parameters are not explained.

xukuanHIT, 2024-01-15:
@zhongxiangrong Hi, some parameter explanations can be found [here](https://entuedu-my.sharepoint.com/:u:/g/personal/kuan_xu_staff_main_ntu_edu_sg/EaApKYmW2IFBtG9PhZpF6DgBwXnsp-61gAM49bpE0Zfqpw?e=tNifjM). it's true that LSD is not robust enough and needs to be carefully tuned. A new version with better line detection, loop detection and map reuse will be released soon.

zhongxiangrong, 2024-01-16:
thanks！keyframe: max_angle: 0.5 。What is the unit of this parameter?

zhongxiangrong, 2024-01-16:
When is the new version expected to be released? Have you considered using an IMU sensor?

zhongxiangrong, 2024-01-16:
What's more, What is the unit of parameter "line_detector:  angle_thr,distance_thr, ep_thr"?In addition, it seems that these three parameters are not used in the code, but have been determined.

xukuanHIT, 2024-01-16:
> thanks！keyframe: max_angle: 0.5 。What is the unit of this parameter?

radian

xukuanHIT, 2024-01-16:
> When is the new version expected to be released? Have you considered using an IMU sensor?

may be 3-6 months later. IMU sensor will be supported.

xukuanHIT, 2024-01-16:
> What's more, What is the unit of parameter "line_detector: angle_thr,distance_thr, ep_thr"?In addition, it seems that these three parameters are not used in the code, but have been determined.

radian and pixel. Yes, I give specific parameters directly in the source code.

zhongxiangrong, 2024-01-18:
I found from the keyframe txt file output by the algorithm that the initial position is (0,0,1), why not (0,0,0)?

xukuanHIT, 2024-01-18:
Just for visualization. The initial pose can be set [here](https://github.com/sair-lab/AirVO/blob/796eecd1e2451d2fba9aab6bbb74ed5f70e6957f/src/map_builder.cc#L258).

zhongxiangrong, 2024-01-25:
if(d_angle > angle_thr){
        if(std::abs(angle1) < (M_PI_2 - angle_threshold)){
          break;
        }else{
          continue;
        }
      }
Here is part of the LineDetector::MergeLines function. Should the second judgment statement be greater ?

closed 2024-03-18 by xukuanHIT