sair-lab/AirSLAM #96: 你好，当我跑kitti数据集时出现以下情况，能帮忙看一下吗

NirvanaDragon, 2024-03-02:
[image]
右边没有任何可视化的轨迹，最后也没有提取出任何关键帧
[image]
我跑的是kitti 00 数据集，我根据euroc.yaml文件修改了image_height: 376
image_width: 1241 bf: 386.1448 以及左右目相机的内参，但是只能提取特征点不能进行数据关联，这是怎么回事呢

xukuanHIT, 2024-03-04:
@NirvanaDragon 你好，[这个文件](https://github.com/sair-lab/AirVO/blob/796eecd1e2451d2fba9aab6bbb74ed5f70e6957f/configs/configs_oivio.yaml#L14)里的参数也要修改一下

NirvanaDragon, 2024-03-06:
您好，有做过KITTI数据集的实验吗，这个参数该怎么修改呢，有没有kitti的yaml文件分享一下

xukuanHIT, 2024-03-09:
@NirvanaDragon 改一下[这里](https://github.com/sair-lab/AirVO/blob/796eecd1e2451d2fba9aab6bbb74ed5f70e6957f/configs/configs_euroc.yaml#L15)的图像尺寸大小。[这个文件](https://github.com/sair-lab/AirVO/blob/796eecd1e2451d2fba9aab6bbb74ed5f70e6957f/configs/euroc.yaml)的参数是opencv stereoRectify()函数的输出

NirvanaDragon, 2024-03-09:
您好，我修改了对应的参数，还是不能正常显示轨迹，关键帧的数量依旧是0，只能提取特征点，不能提取线特征

xukuanHIT, 2024-03-09:
Tracking lost的时候不会显示线特征。能把去完畸变后的图像show出来吗？

NirvanaDragon, 2024-03-11:
[image]
你好，这是我去完畸变之后的图像

xukuanHIT, 2024-03-16:
@NirvanaDragon 你好，我是来下kitti数据集。kitti的双目已经做过双目校正了，但图像分辨率是1226*370的，宽和高都不能被8整除。superpoint在的feature map最小是原图的1/8,所以直接用原图推理会有问题，你看你图像上的特征点的位置根本就不对，不在该提取特征的地方，这导致tracking失败进而程序没运行到提取线特征这一步。
一个简单的解决办法是crop图片左上角的1224*368区域，一定要是左上角，这样图像内参也不用更改，宽和高都能被8整除

NirvanaDragon, 2024-03-18:
`try
    {
        cv_ptrLeft = cv_bridge::toCvShare(imgLeft);
        cv_ptrRight = cv_bridge::toCvShare(imgRight);
        // 定义裁剪区域的位置和尺寸
        int cropX = 0;        // 裁剪区域的左上角X坐标
        int cropY = 0;        // 裁剪区域的左上角Y坐标
        int cropWidth = 1224; // 裁剪区域的宽度
        int cropHeight = 368; // 裁剪区域的高度

        // 裁剪左目图像
        croppedLeft = cv_ptrLeft->image(cv::Rect(cropX, cropY, cropWidth, cropHeight));

        // 裁剪右目图像
        croppedRight = cv_ptrRight->image(cv::Rect(cropX, cropY, cropWidth, cropHeight));
    }
    catch (cv_bridge::Exception &e)
    {
        ROS_ERROR("cv_bridge exception: %s", e.what());
        return;
    }

    static int frame_id = 0;
    auto before_infer = std::chrono::steady_clock::now();

    InputDataPtr input_data = std::shared_ptr<InputData>(new InputData());
    input_data->index = frame_id;
    input_data->image_left = croppedLeft.clone();
    input_data->image_right = croppedRight.clone();
    input_data->time = imgLeft->header.stamp.toSec();

    if (input_data == nullptr)
        return;
    p_map_builder->AddInput(input_data);`
    你好，我按照你说的方式修改了数据的尺寸，但是还是不能正常运行，您能看看我修改的对吗

xukuanHIT, 2024-03-18:
是的。相应配置文件里的图像分辨率也要改成1224和368

NirvanaDragon, 2024-03-18:
[image]
[image]
[image]
您好，我已经将配置文件中的分辨率都修改了，但是还是不能正常运行，您知道原因吗，不知道您那里是否可以正常运行kitti数据集

xukuanHIT, 2024-03-18:
我试了下是可以的。看着你给的图片superpoint提取成功了，应该superglue没匹配上。可以把原来生成的superglue的.engine删掉再运行下试试

NirvanaDragon, 2024-03-18:
你好，我将原来的superglue删除了，重新生成了一份，但是仍然没有匹配成功，我用的是superglue_outdoor_sim_int32.engine/(ㄒoㄒ)/~~

xukuanHIT, 2024-03-18:
[kitti_config.zip](https://github.com/sair-lab/AirVO/files/14635135/kitti_config.zip)
你试试这俩配置文件

NirvanaDragon, 2024-03-18:
您好，你运行的是kitti的哪种类型是数据集呢，是对原始odometry数据集录的bag吗，还是kitti的extract或者syn的数据集呢

xukuanHIT, 2024-03-18:
用的kitti odometry双目图像序列，第“05”个

NirvanaDragon, 2024-04-08:
好吧，不能匹配的原因在于superglue不适配我的tensorrt的版本

closed 2024-04-26 by xukuanHIT