sair-lab/AirSLAM #97: How to Evaluate VO?

ZhangshenHuang, 2024-03-10:
Thank you for your open-source contribution. I have two questions: 
1.How to evaluate the performance of VO, such as the tables listed in your paper, for example, RMSE metrics and the running time of each section? 
2.When will AirVO be extended to a full SLAM system?

xukuanHIT, 2024-03-16:
@ZhangshenHuang

xukuanHIT, 2024-03-16:
Hi @ZhangshenHuang 
1. I'm not entirely clear on your first question. Are you asking how to get the values of these metrics or what kinds of metrics should be used to evaluate a VO?
2. We are doing experiments and writing the paper. May after about three more months we can open-source the code.

ZhangshenHuang, 2024-03-17:
Thank you for your reply. For example, after running under the Euroc dataset, a traj.txt file will be generated. What do the data in the file represent? How can standard values and trajectory images from the paper be obtained from the traj.txt file?

xukuanHIT, 2024-03-18:
Yes, evo.

closed 2024-04-26 by xukuanHIT