sair-lab/AirSLAM #100: buffers.h文件中的问题

Cai-RS, 2024-03-23:
你好，感谢你的工作！
我想问一下，在buffers.h文件中，在host上分配的用于数据传输的内存，是否使用页锁定内存更好呢？即是否应该使用cudaMallocHost()/cudaFreeHost()，而不是分配分页内存的Malloc()/free()？
[image]

xukuanHIT, 2024-04-13:
@yuefanhao

yuefanhao, 2024-04-13:
@Cai-RS I refer to the implementation in the official sample code of TensorRT [https://github.com/NVIDIA/TensorRT/blob/28733f0fdccde2967fed395b06ca491af3a561a9/samples/common/buffers.h#L202](https://github.com/NVIDIA/TensorRT/blob/28733f0fdccde2967fed395b06ca491af3a561a9/samples/common/buffers.h#L202),  in addition, here is an explanation of the difference between cudaMallocHost and malloc [https://stackoverflow.com/questions/23133203/cudamallochost-vs-malloc-for-better-performance-shows-no-difference](https://stackoverflow.com/questions/23133203/cudamallochost-vs-malloc-for-better-performance-shows-no-difference).

closed 2024-04-26 by xukuanHIT