sair-lab/AirSLAM #101: Error when runining the examples

visionbike, 2024-03-27:
Hi,

Thank you for sharing code. I tried to run the AirVO code but meet the following problems:

One Frame Processinh Time: 17 ms.
Save file to /home/asrlab/catkin_ws/src/AirVO/debug/traj.txt
_keyframe_ids.size = 155
terminate called without an active exception
[air_vo-2] process has died [pid 148170, exit code -6, cmd /home/asrlab/catkin_ws/devel/lib/air_vo/air_vo __name:=air_vo __log:=/home/asrlab/.ros/log/df79dbc6-ec4a-11ee-b1c8-5b0f24582382/air_vo-2.log].
log file: /home/asrlab/.ros/log/df79dbc6-ec4a-11ee-b1c8-5b0f24582382/air_vo-2*.log

My system:
OS: Ubuntu 20.04
GPU: RTX 3070
CUDA: 11.8
cuDNN: 8.5
OpenCV: 4.2.0
I compiled OpenCV 4.2.0 with C++14 and CUDA.

xukuanHIT, 2024-04-13:
@visionbike Hi, the program is over and the trajectory has been saved successfully. The error is caused by the visualization thread. 
 It doesn't affact the results and you can ignore it. The bug will be fixed in the next version.

visionbike, 2024-04-18:
Thank you for your answer! I've already realized this. Looking forward to your next version soon.

closed 2024-04-18 by visionbike

ztx991028, 2024-05-28:
运行完为什么找不到保存的轨迹？debug没有这个文件夹。谢谢

xukuanHIT, 2024-05-28:
@ztx991028 可以创建一个debug文件夹，或者把保存路径设置为任意一个已经存在的文件夹

ztx991028, 2024-05-28:
> @ztx991028可以创建一个debug文件夹，或者把保存路径设置为任意一个已经存在的文件夹

谢谢！

ztx991028, 2024-05-28:
> @ztx991028可以创建一个debug文件夹，或者把保存路径设置为任意一个已经存在的文件夹

我已经得到轨迹了,但它只有101行，这是正常的吗？真值有7000多行，该怎样和真值进行评估？期待你的回复，谢谢

xukuanHIT, 2024-05-28:
@xukuanHIT 只保存了keyframe的轨迹，可以检查下时间戳，如果没少很多的话应该就是正常的。评估的话用evo就可以