sair-lab/AirSLAM #104: Regarding computer configuration requirements

Lmy971109, 2024-04-21:
Dear author:
  Thank you so much for the great work you do for VO! Excuse me, does the configuration of this computer meet the environment configuration of AirVO? It is using dual system Ubuntu18.04. The ROS version is melodic and CUDA is 11.3.
[image]

xukuanHIT, 2024-04-21:
@Lmy971109 Hi, the GPU is ok, but we have not tried AirVO with CUDA 11.3. And what is the driver version on your computer?

Lmy971109, 2024-04-21:
Dear author:
   The drivers installed on this computer are as follows:
[image]

xukuanHIT, 2024-04-22:
The driver is ok. You can try the docker first

Lmy971109, 2024-04-22:
OK,Thanks very much

---- Replied Message ----
| From | Xu ***@***.***> |
| Date | 04/22/2024 18:44 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [sair-lab/AirVO] Regarding computer configuration requirements (Issue #104) |

The driver is ok. You can try the docker first

closed 2024-05-06 by xukuanHIT