sair-lab/AirSLAM #106: Pose file predicted by PL-SLAM

WYKXD, 2024-04-25:
Dear author:
     Thank you so much for the great work you have done for VO! I want to compare PL-SLAM with other visual odometers. Can you provide the pose file predicted by PL-SLAM on the UMA-VI dataset? My E-mail address is: acadscill@gmail.com

xukuanHIT, 2024-05-06:
@WYKXD Hi, it is recommended to run PLSLAM on your computer for a fair comparison.

WYKXD, 2024-05-07:
@xukuanHIT Hello, thank you very much for your reply. My computer configuration is the same as what you mentioned in your previous comment. I cannot download the UMA-VI data set. I can refer to your article for the ATE results of PL-SLAM on the UMA-VI dataset, but I don't have the trajectory file of PL-SLAM predictions in the visual analysis. I just need the track files of the four sequences of PL-SLAM in your article. Can you help me?

xukuanHIT, 2024-05-10:
@WYKXD Hi, the official download link has been recovered now. Besides, I have uploaded the sequences used in the paper, please refer to #70 . You can download the dataset through either of them.