sair-lab/AirSLAM #108: Runtime Analysis

Wu-ZW, 2024-05-15:
when I run euroc datasets, some parameters as follows: image size is set 640*480, superpoint max_keypoints is set 200,
[image]
the time of detecting and tracking feature points for one image is about 92 ms, more than 65 ms in paper. Is any parameters i need to adjust?

Wu-ZW, 2024-05-15:
l used Orin NX (16GB) edition.

xukuanHIT, 2024-05-15:
@Wu-ZW HI, we run on the  Nvidia Jetson AGX Xavier for runtime analysis. The computing capability of Jetson AGX Xavier is better than your  Orin NX.

Wu-ZW, 2024-05-15:
thanks for your replay! emmm,  the computing capability of jeston orin nx (8.7) is larger than AGX Xavier(7.2).
[image]

xukuanHIT, 2024-05-16:
Sorry for my mistake. The comparison can be found [here](https://developer.nvidia.com/embedded/jetson-modules#:~:text=TX2%20NX-,AI%20Performance,-275%20TOPS), but it is true that jeston orin nx is  more powerful than AGX Xavier. 
Can you tell me your version of TensorRT? and have you tried the original Python version of superpoint and superglue on your jeston orin nx?

Wu-ZW, 2024-05-16:
I have run original code of sp and  sg using PyTorch-CUDA, it is Inefficient.
the consumption of point extraction and point matching is unstable, 
the parameters as follow:
[image]
tensorrt + CUDA ....version as follows:
[image]

xukuanHIT, 2024-05-16:
Can you compare the average runtime of C++ and Python code? On our Jetson platform, the C++ feature extraction and matching is about  6x faster than the Python version. I think if a similar improvement can be achieved on your Jetson, we can conclude that Orin NX is actually not as good as AGX Xavier.

Wu-ZW, 2024-05-16:
parameters:
max_keypoints:200
keypoint_threshold:0.004
match_threshold:0.1
Python (Pytorch-CUDA) Code:
Superpoint time: 0.10092282295227051
Superglue time: 0.10480713844299316
C++ TensorRT Code:
sp:adout 40 ms
sg:about 50 ms

Wu-ZW, 2024-05-16:
In jeston Orin NX， the C++ feature extraction and matching is about 2x faster than the Python version. the improvement is not very obvious.

xukuanHIT, 2024-05-16:
It's strange. I am not sure whether it is also caused by the TensorRT version. We have also encountered the efficiency problem when using TensorRT 8.5 on GeForce RTX GPUs that are lower than 40 series.  It is recommended to try to downgrade the driver version to 8.4.1.5 if possible.

xukuanHIT, 2024-05-16:
May be similar this [issue](https://github.com/yuefanhao/SuperPoint-SuperGlue-TensorRT/issues/32).

Wu-ZW, 2024-05-16:
thanks a lot. I am also confused about above results. Addition, could I ask a question about how to export the PyTorch model to ONNX (int64)? I saw the scripts of convert_int32.py, is it convert INT64 to INT32 or FLOAT 32 to INT32?

Wu-ZW, 2024-05-16:
> May be similar to this [issue](https://github.com/yuefanhao/SuperPoint-SuperGlue-TensorRT/issues/32).
I doubt that "read image again", the same code runs normal on other platform.

xukuanHIT, 2024-05-20:
> thanks a lot. I am also confused about above results. Addition, could I ask a question about how to export the PyTorch model to ONNX (int64)? I saw the scripts of convert_int32.py, is it convert INT64 to INT32 or FLOAT 32 to INT32?

Yes, just run the script.

xukuanHIT, 2024-05-20:
> > May be similar to this [issue](https://github.com/yuefanhao/SuperPoint-SuperGlue-TensorRT/issues/32).
> > I doubt that "read image again", the same code runs normal on other platform.

I mean the different environments may affect the runtime.

closed 2024-08-12 by xukuanHIT