sair-lab/AirSLAM #112: Failed when extracting point features !

cpymaple, 2024-07-03:
Hi, Thanks for your great work!

I have successfully compile and run AirVO in the docker environment. But I still want to configure the dependencies in my ubuntu. I compile the AirVO in my ubuntu without any error. However, when I run the system, the terminal show **"Failed when extracting point features !"**
[image]

Here are my dependencies: OpenCV 4.2, Eigen 3.3.9, Ceres 2.0.0, G2O (20230223), TensorRT 8.4.1.5.(Tar file), CUDA 11.6, cudnn.8.5(Tar file), python3.8, onnx, Ros noetic, boost, glog, nvidia-driver-535. And I also export the following codes in my ~/.bashrc: source ~/catkin_airvo/devel/setup.bash; export LD_LIBRARY_PATH=/home/cpy/TensorRT-8.4.1.5/lib:$LD_LIBRARY_PATH; export LD_LIBRARY_PATH=/usr/local/cuda/lib64:$LD_LIBRARY_PATH.

By the way, in order to find the .h file in TensorRT (/home/cpy/TensorRT-8.4.1.5) and successfully compile AirVO, I add **include_directories( /home/cpy/TensorRT-8.4.1.5/include)**; and **link_directories(/home/cpy/TensorRT-8.4.1.5/lib)** in CMakeList. But when I use the docker env, I do not need to revise the CMakeList.

Is the problem may related to my TensorRT and nvidia-driver? SInce the system can run and only cannot extract features, maybe the superpoint do not successfully operate? But for all dependencies, I follow the official tutorial, it should be correct. And I also follow other similar issues in this repo, but those does not work for me.

Looking forward to your kind reply~

cpymaple, 2024-07-03:
Update: 1. My computer uses the RTX965M GPU.  2.I found that after successfully compile AirVO under docker env, I only delete build/ devel/ folder. I notice that I also need to delete the .engine file. After that, When I firstly run the AirVO, the system shows the following output:
```setting /run_id to f06f9c2a-394a-11ef-944a-3b7ef229889a
process[rosout-1]: started with pid [22195]
started core service [/rosout]
process[air_vo-2]: started with pid [22202]
config_file = /home/cpy/catkin_airvo/src/AirVO/configs/configs_euroc.yaml
path = /home/cpy/Datasets/MH_01_easy/cam0/data
[07/03/2024-22:45:50] [I] [TRT] [MemUsageChange] Init CUDA: CPU +197, GPU +0, now: CPU 224, GPU 590 (MiB)
[07/03/2024-22:45:52] [I] [TRT] [MemUsageChange] Init builder kernel library: CPU +6, GPU +1, now: CPU 249, GPU 591 (MiB)
[07/03/2024-22:45:52] [I] [TRT] ----------------------------------------------------------------
[07/03/2024-22:45:52] [I] [TRT] Input filename:   /home/cpy/catkin_airvo/src/AirVO/output/superpoint_v1_sim_int32.onnx
[07/03/2024-22:45:52] [I] [TRT] ONNX IR version:  0.0.8
[07/03/2024-22:45:52] [I] [TRT] Opset version:    12
[07/03/2024-22:45:52] [I] [TRT] Producer name:    onnx-typecast
[07/03/2024-22:45:52] [I] [TRT] Producer version: 
[07/03/2024-22:45:52] [I] [TRT] Domain:           
[07/03/2024-22:45:52] [I] [TRT] Model version:    0
[07/03/2024-22:45:52] [I] [TRT] Doc string:       
[07/03/2024-22:45:52] [I] [TRT] ----------------------------------------------------------------
[07/03/2024-22:45:52] [W] [TRT] onnx2trt_utils.cpp:369: Your ONNX model has been generated with INT64 weights, while TensorRT does not natively support INT64. Attempting to cast down to INT32.
[07/03/2024-22:45:52] [W] [TRT] FP16 support requested on hardware without native FP16 support, performance will be negatively affected.
[07/03/2024-22:46:01] [I] [TRT] [MemUsageChange] Init cuBLAS/cuBLASLt: CPU +259, GPU +100, now: CPU 516, GPU 691 (MiB)
[07/03/2024-22:46:02] [I] [TRT] [MemUsageChange] Init cuDNN: CPU +81, GPU +10, now: CPU 597, GPU 701 (MiB)
[07/03/2024-22:46:02] [I] [TRT] Local timing cache in use. Profiling results in this builder pass will not be stored.
[07/03/2024-22:46:13] [W] [TRT] Weights [name=Conv_0 + Relu_1.weight] had the following issues when converted to FP16:
[07/03/2024-22:46:13] [W] [TRT]  - Subnormal FP16 values detected. 
[07/03/2024-22:46:13] [W] [TRT] If this is not the desired behavior, please modify the weights or retrain with regularization to reduce the magnitude of the weights.
[07/03/2024-22:46:13] [I] [TRT] Some tactics do not have sufficient workspace memory to run. Increasing workspace size will enable more tactics, please check verbose output for requested sizes.
[07/03/2024-22:46:13] [W] [TRT] Weights [name=Conv_0 + Relu_1.weight] had the following issues when converted to FP16:
[07/03/2024-22:46:13] [W] [TRT]  - Subnormal FP16 values detected. 
[07/03/2024-22:46:13] [W] [TRT] If this is not the desired behavior, please modify the weights or retrain with regularization to reduce the magnitude of the weights.
[07/03/2024-22:46:16] [W] [TRT] Weights [name=Conv_2 + Relu_3.weight] had the following issues when converted to FP16:
[07/03/2024-22:46:16] [W] [TRT]  - Subnormal FP16 values detected. 
[07/03/2024-22:46:16] [W] [TRT]  - Values less than smallest positive FP16 Subnormal value detected. Converting to FP16 minimum subnormalized value. 
[07/03/2024-22:46:16] [W] [TRT] If this is not the desired behavior, please modify the weights or retrain with regularization to reduce the magnitude of the weights.
[07/03/2024-22:46:16] [W] [TRT] Weights [name=Conv_2 + Relu_3.weight] had the following issues when converted to FP16:
[07/03/2024-22:46:16] [W] [TRT]  - Subnormal FP16 values detected. 
[07/03/2024-22:46:16] [W] [TRT]  - Values less than smallest positive FP16 Subnormal value detected. Converting to FP16 minimum subnormalized value. 
[07/03/2024-22:46:16] [W] [TRT] If this is not the desired behavior, please modify the weights or retrain with regularization to reduce the magnitude of the weights.
[07/03/2024-22:46:17] [W] [TRT] Weights [name=Conv_5 + Relu_6.weight] had the following issues when converted to FP16:
[07/03/2024-22:46:17] [W] [TRT]  - Subnormal FP16 values detected. 
[07/03/2024-22:46:17] [W] [TRT]  - Values less than smallest positive FP16 Subnormal value detected. Converting to FP16 minimum subnormalized value. 
[07/03/2024-22:46:17] [W] [TRT] If this is not the desired behavior, please modify the weights or retrain with regularization to reduce the magnitude of the weights.
[07/03/2024-22:46:17] [W] [TRT] Weights [name=Conv_5 + Relu_6.weight] had the following issues when converted to FP16:
[07/03/2024-22:46:17] [W] [TRT]  - Subnormal FP16 values detected. 
[07/03/2024-22:46:17] [W] [TRT]  - Values less than smallest positive FP16 Subnormal value detected. Converting to FP16 minimum subnormalized value. 
[07/03/2024-22:46:17] [W] [TRT] If this is not the desired behavior, please modify the weights or retrain with regularization to reduce the magnitude of the weights.
[07/03/2024-22:46:17] [W] [TRT] Weights [name=Conv_7 + Relu_8.weight] had the following issues when converted to FP16:
[07/03/2024-22:46:17] [W] [TRT]  - Subnormal FP16 values detected. 
[07/03/2024-22:46:17] [W] [TRT]  - Values less than smallest positive FP16 Subnormal value detected. Converting to FP16 minimum subnormalized value. 
[... cut ...]
1.Could you please tell me why I can successfully run AirVO in one time under the docker environment, but in my own system environment I need to run three times to get it to work? Is it related to the loading onnx or generating .engine files? 

2.Meanwhile, looking at the log messages which show memory errors and other issues, is this related to the low configuration of my computer? 

3.By the way, the tutorial shows the ceres 2.0.0 is necessary, but I do not find the AirVO use ceres to perform optimization. Do we not actually need ceres, just g2o for optimization?

xukuanHIT, 2024-07-08:
@cpymaple Hi, 

1. The ".engine"  file is generated based on your development environment, so each time you change environments, you need to delete and regenerate it. Besides, generating this file takes some time, so please be patient. Once it is generated, you can proceed to run it.

2. They are outputs of TensorRT. If you can run successfully finally, you can ignore them.

3. We meet some build errors when the versions of the two libraries do not match, which is similar to [this issue](https://github.com/RainerKuemmerle/g2o/issues/477). I think it can also work if Ceres is not installed on your computer.

cpymaple, 2024-07-08:
> @cpymaple Hi,
>
>     1. The ".engine"  file is generated based on your development environment, so each time you change environments, you need to delete and regenerate it. Besides, generating this [...]

Hi, for the questions 3., I also encounter the mentioned problems. Thanks for your kind reply.

closed 2024-07-08 by cpymaple