sair-lab/AirSLAM #113: Ubuntu 20.04 docker enviroment error: unknown or invalid runtime name: nvidia.

Lx017, 2024-07-09:
Hi, I have already pulled the docker successfully (with sudo if that matters) and run:
sudo docker run -it --env DISPLAY=$DISPLAY --volume /tmp/.X11-unix:/tmp/.X11-unix --privileged --runtime nvidia --gpus all --volume ${PWD}:/workspace --workdir /workspace --name air_slam xukuanhit/air_slam:v1 /bin/bash
However, docker gives me:
docker: Error response from daemon: unknown or invalid runtime name: nvidia.
what is the solution?

xukuanHIT, 2024-07-09:
@Lx017 Hi, you need to install the [nvidia docker](https://docs.nvidia.com/datacenter/cloud-native/container-toolkit/latest/install-guide.html).

Lx017, 2024-07-12:
I cannot get configuration and setup right in any way on Ubuntu 20.04, main issue is from nvidia and docker. But 22.04 can work.

closed 2024-07-13 by Lx017