sair-lab/AirSLAM #116: roslaunch euroc.launch ;process has died

whrcn, 2024-07-26:
平台：3060
ubuntu20.04 
opencv 4.2
CUDA 11.8
TensorRT 8.5
ceres2.0.0
g2o(tag:20230223_git)
编译中间有报错但是解决了，我下载的euroc的数据集，运行roslaunch air_vo euroc_ros.launch后显示如下：
Save file to /home/hrw/catkin_airvio/src/AirVO/debug/traj.txt
_keyframe_ids.size = 154
terminate called without an active exception
[air_vo-2] process has died [pid 80377, exit code -6, cmd /home/hrw/catkin_airvio/devel/lib/air_vo/air_vo __name:=air_vo __log:=/home/hrw/.ros/log/604265a2-4b28-11ef-b206-a7b418734395/air_vo-2.log].
log file: /home/hrw/.ros/log/604265a2-4b28-11ef-b206-a7b418734395/air_vo-2*.log

[image]
也没有可视化界面，请问这是怎么回事呢？

whrcn, 2024-07-26:
当我在运行UMA-VI数据集的时候，也是类似的报错：One Frame Processinh Time: 30 ms.
Save file to /home/hrw/catkin_airvio/src/AirVO/debug/traj.txt
_keyframe_ids.size = 305
terminate called without an active exception
[air_vo-2] process has died [pid 89162, exit code -6, cmd /home/hrw/catkin_airvio/devel/lib/air_vo/air_vo __name:=air_vo __log:=/home/hrw/.ros/log/8e3f65c0-4b2a-11ef-b206-a7b418734395/air_vo-2.log].
log file: /home/hrw/.ros/log/8e3f65c0-4b2a-11ef-b206-a7b418734395/air_vo-2*.log
也是类似的报错
[image]
，但是在读取了1000多张图片的时候就中断了

xukuanHIT, 2024-07-26:
@whrcn 你好，
1. 程序里只发布了可视化相关的topic，并没有默认打开rviz，你可以手动打开rviz可视化这些topic。
2. 这个报错是可视化多线程导致的，程序结束时这些线程并没有妥善停止，不过这个不影响程序运行，因为这是程序结束时发生的，轨迹也已经保存。这个bug在新版已经修复了，但目前还没开源。
3. UMA这个程序也运行结束了，并且轨迹也保存了。你可以检查下输入数据，是否只有1176帧？

whrcn, 2024-07-29:
非常感谢，已经解决了这个问题！这个中断是因为输入结束了而不是异常。