sair-lab/AirSLAM #122: Little question about the  [TABLE II] in the paper .

GeYapeng, 2024-08-14:
Thank you for the great work of AirSLAM .
I'm a little confused when I read the paper . Here in the [TABLE II] when comparing with ORB-SLAM3 with loop , it shows the stereo-inertial(SI)  data of [0.036,  0.033,  0.035 ... ] , but marked with monocular-inertial(MI) .
There might be some misunderstanding .Which group of data should I refer to ? Could you please give me some advice ?
Looking forward to your reply ~

xukuanHIT, 2024-08-14:
@GeYapeng Hi, thanks for the reminder. This should be the result for Stereo Inertial ORB-SLAM3. We've updated it in the paper, and the revised version will be released recently.

GeYapeng, 2024-08-14:
Thanks !  Got it ~  \ >_< /

closed 2024-08-16 by xukuanHIT