sair-lab/AirSLAM #123: Model conversion error

aax1n, 2024-08-14:
Hello, thank you for open-sourcing such a great project. May I ask what are the specific version numbers of the libraries used, such as CUDA and TensorRT? I've tried many version combinations with CUDA 12 and TensorRT 8.6, but there are always issues with model conversion.

xukuanHIT, 2024-08-14:
@aax1n Hi, we use Cuda 12.1 and TensorRT 8.6.1.6. The README has been updated.

aax1n, 2024-08-16:
> @aax1n您好，我们使用 Cuda 12.1 和 TensorRT 8.6.1.6。README 已更新。

I followed your suggestion and switched to CUDA 12.1 and TensorRT 8.6.1.1 for testing, but the following error keeps occurring.

[image]

xukuanHIT, 2024-08-16:
@yuefanhao Could you take a look at this issue?

aax1n, 2024-08-16:
@xukuanHIT It was my own issue. I have multiple versions of the TensorRT library, and Mod is only supported in versions 8.5 and above. I didn't refresh the previous environment variables before compiling, which caused the error.

xukuanHIT, 2024-08-21:
OK