sair-lab/AirSLAM #125: runtime_error: Time is out of dual 32-bit range

Ling-Bao, 2024-08-14:
Hello, thank you for open-sourcing such a great project. I meet a problem but unable to locate the erroneous codes. 
-----------------------------------------------
Error info as follow:
terminate called after throwing an instance of 'std::runtime_error'
  what():  Time is out of dual 32-bit range

Running environments: 
(1) docker xukuanhit/air_slam:v4
(2) custom datasets and check correct, 1~7000+ images running correct, but around 70000 images meet this problem and each times not fix frame id
-----------------------------------------------

How to troubleshoot this problem? Do you have any good suggestions?

xukuanHIT, 2024-08-15:
@Ling-Bao Hi, is your data named with timestamps?  And can you provide more output or log information?

Ling-Bao, 2024-08-15:
> @Ling-Bao Hi, is your data named with timestamps? And can you provide more output or log information?

Thanks replay, 
Custom datasets is Euroc format, and I check timestamp of image and imu.
No any other error information, program just terminal

xukuanHIT, 2024-08-15:
@Ling-Bao I think the best way is to use some debugging techniques to identify where the program exits. If that is not easy, you could try a few things first:
1. Run the system in pure stereo mode to check if the issue is with the image data or the IMU data.
2. Start from the 7,000th image to see if the program still stops at the same image, which will help find if the issue is related to a specific image or the total number of images.

Ling-Bao, 2024-08-15:
> @Ling-Bao I think the best way is to use some debugging techniques to identify where the program exits. If that is not easy, you could try a few things first:
>
> 1. Run the system in pure stereo mode [...]

Yes, suggestion is helpful, I plan to debugging. 
By the way, 1 & 2  is carefully test, and found a rule when running more than 7000 images will meet this error. I guess it's maybe related to long-term running?

xukuanHIT, 2024-08-15:
@Ling-Bao We have tested it with more than 10,000 images (the mapping demo in our video) and it works well.

closed 2024-10-28 by xukuanHIT