sair-lab/AirSLAM #127: 启动之后没有任何反应

silence-moon, 2024-08-16:
系统启动起来之后，无图像也没有建图的输出。若是改变模型路径，则会提示找不到模型，说明应该是在运行，但是具体卡在哪里了，不太清楚
[image]

xukuanHIT, 2024-08-16:
@silence-moon 输出信息提示你的这个[配置文件](https://github.com/sair-lab/AirSLAM/blob/master/configs/camera/euroc.yaml)没找到，你可以检查下是否存在

silence-moon, 2024-08-16:
不好意思，是下面这一张
[image]

silence-moon, 2024-08-16:
哦，看到输出了，不支持双目。请问我需要怎么修改呢？

xukuanHIT, 2024-08-16:
支持双目。第一次运行会根据.onnx生成.enginew文件，这个步骤比较耗时，需要等几分钟完成模型转换

silence-moon, 2024-08-16:
然后，刚刚突然多了很多输出，就又报错了
[image]

silence-moon, 2024-08-16:
当我再次启动之后，输出是这样子的
[image]

silence-moon, 2024-08-16:
嗯，这里的launch文件，我按照readme中mapping的说明做了设置，<arg name="dataroot" default = "/home/silence/SLAM/datasets/MH_01/mav0/" />，saving_dir也是存在的

xukuanHIT, 2024-08-16:
@silence-moon 你测试 环境是怎样的？Cuda和TenrorRT版本是多少呢？

silence-moon, 2024-08-16:
我的cuda版本是11.4，tensorrt的版本是TensorRT-8.6.0.12.Linux.x86_64-gnu.cuda-11.8.tar.gz

silence-moon, 2024-08-16:
我准备先在命令行中直接转换engine文件，但是有部分文件执行错误了，部分文件转换成功了。
总的执行结果图是这样的：
[image]

错误的plnet模型报错如下：
[image]

silence-moon, 2024-08-16:
它可以使用我转换好的模型吗？刚才我抓换了之后，启动roslaunch的时候，代码内部又执行了一遍，但是我还没有在代码中找到进行转换的代码。如果是代码中执行的话，转换会报错，如下：
[E] [TRT] 3: [runtime.cpp::~Runtime::346] Error Code 3: API Usage Error (Parameter check failed at: runtime/rt/runtime.cpp::~Runtime::346, condition: mE
ngineCounter.use_count() == 1. Destroying a runtime before destroying deserialized engines created by the runtime leads to undefined behavior.                                
) 
是不是因为，模型没有转换成功，所以后面才报错：
[visual_odometry-2] process has died [pid 3339556, exit code -11,
然后，执行不下去的呢？

xukuanHIT, 2024-08-16:
@silence-moon 理论上是可以使用转好的模型的，但你好像没有把plnet的模型转好。你可以先用docker试试？

silence-moon, 2024-08-19:
我下载了和您一样的tensorrt版本，但是直接运行起来还是报错，以下是转换日志：

[08/19/2024-11:52:26] [W] [TRT] onnx2trt_utils.cpp:374: Your ONNX model has been generated with INT64 weights, while TensorRT does not natively support INT64. Attempting to cast down to INT32.
[08/19/2024-11:52:26] [W] [TRT] onnx2trt_utils.cpp:400: One or more weights outside the range of INT32 was clamped
[08/19/2024-11:52:26] [W] [TRT] Tensor DataType is determined at build time for tensors not marked as input or output.
[08/19/2024-11:52:26] [W] [TRT] Tensor DataType is determined at build time for tensors not marked as input or output.
[08/19/2024-11:52:26] [I] Finished parsing network model. Parse time: 0.0948283
[08/19/2024-11:52:26] [W] Dynamic dimensions required for input: input, but no shapes were provided. Automatically overriding shape to: 1x1x1x1
[08/19/2024-11:52:26] [I] [TRT] BuilderFlag::kTF32 is set but hardware does not support TF32. Disabling TF32.
[08/19/2024-11:52:26] [E] Error[4]: [graphShapeAnalyzer.cpp::processCheck::862] Error Code 4: Internal Error (/TopK: length of reduction axis (0) is smaller than K (300))
[08/19/2024-11:52:26] [E] Engine could not be created from network
[08/19/2024-11:52:26] [E] Building engine failed
[08/19/2024-11:52:26] [E] Failed to create engine from model or file.
[08/19/2024-11:52:26] [E] Engine set up failed
&&&& FAILED TensorRT.trtexec [TensorRT v8601] # /home/silence/software/TensorRT-8.6.1.6/bin/trtexec --onnx=plnet_s0.onnx --saveEngine=plnet_s0.engine

最后一行是转换的命令。麻烦您帮忙看一下，谢谢！

xukuanHIT, 2024-08-19:
我们用的Cuda12.1，我不确定你这个没转成功有没有cuda版本不同的因素在里头。并且，TensorRT 8.6能否与你的Cuda11.4兼容（参考这个[link](https://docs.nvidia.com/deeplearning/tensorrt/release-notes/index.html#rel-8-6-1)）？你能跑同TensorRT给的一些demo吗？

silence-moon, 2024-08-19:
之前我把yolov5转换成tensorrt模型进行了测试，所以跑通应该所没有问题的

xukuanHIT, 2024-08-19:
@yuefanhao 越凡，你可以看下这个问题吗？

yuefanhao, 2024-08-20:
@silence-moon @xukuanHIT 我看了下问题记录，模型转换的过程中我看到这行报错[08/19/2024-11:52:26] [E] Error[4]: [graphShapeAnalyzer.cpp::processCheck::862] Error Code 4: Internal Error (/TopK: length of reduction axis (0) is smaller than K (300))，手动模型转换的时候指定动态尺寸了吗？但如果使用我们提供的模型也无法成功运行的话，建议还是在我们提供的docker里面尝试下，有可能是cuda版本差异导致的。

silence-moon, 2024-08-20:
没有指定动态尺寸。我转换的时候命令是这样的：trtexec --onnx=*.onnx --saveEngine=*.engine，请问需要加哪些参数啊？

yuefanhao, 2024-08-20:
@silence-moon 可以参考plnet.cpp里面的这两块的代码，https://github.com/sair-lab/AirSLAM/blob/b50b261c9992f68ea9af1f0592fda5d0ac92b458/src/plnet.cpp#L73和https://github.com/sair-lab/AirSLAM/blob/b50b261c9992f68ea9af1f0592fda5d0ac92b458/src/plnet.cpp#L130，是指定哪些输入是动态的；TensorRT官方例程里面有指定动态维度的方法，参考https://github.com/NVIDIA/TensorRT/tree/main/samples/trtexec里面的Example 3

logic-zhang, 2024-08-26:
> @silence-moon @xukuanHIT 我看了下问题记录，模型转换的过程中我看到这行报错[08/19/2024-11:52:26] [E] Error[4]: [graphShapeAnalyzer.cpp::processCheck::862] Error Code 4: Internal Error (/TopK: length of reduction axis (0) is smaller than K (300))，手动模型转换的时候指定动态尺寸了吗？但如果使用我们提供的模型也无法成功运行的话，建议还是在我们提供的docker里面尝试下，有可能是cuda版本差异导致的。

HI,请问AirSLAM支持cuda11.4吗？我的电脑对应的CUDA Version: 11.4，在使用你们的docker环境运行时出现了错误：
/workspace/air_slam_ws/src/AirSLAM/configs/visual_odometry/vo_euroc.yaml
config done
Erron lightglue building
[08/25/2024-17:21:53] [W] [TRT] Unable to determine GPU memory usage
[08/25/2024-17:21:53] [W] [TRT] Unable to determine GPU memory usage
[08/25/2024-17:21:53] [I] [TRT] [MemUsageChange] Init CUDA: CPU +0, GPU +0, now: CPU 24, GPU 0 (MiB)
[08/25/2024-17:21:53] [W] [TRT] CUDA initialization failure with error: 804. Please check your CUDA installation:  http://docs.nvidia.com/cuda/cuda-installation-guide-linux/index.html
Error in SuperPoint building

xukuanHIT, 2024-08-29:
@logic-zhang 提供的docker里面有cuda 12，对机子的nvidia driver版本有要求

closed 2024-10-10 by xukuanHIT

kinggreat24, 2025-02-08:
> 但是直接运行起来还是报错，以下是转
请问你解决这个问题了吗，我也是同样的问题，用的cuda-12.1 tensorRT8.6.1.6版本

silence-moon, 2025-02-08:
我这边应该是启动是的tensorrt模型没有转成功的原因

kinggreat24, 2025-02-08:
> 我这边应该是启动是的tensorrt模型没有转成功的原因
> […](#)
> 
> ---原始邮件---
> 发件人: ***@***.***&gt;
> 发送时间: 2025年2月8日(周六) 晚上7:16
> 收件人: ***@***.***&gt;;
> 抄送: ***@***.***&gt;;"State ***@***.***&gt;;
> 
> 
> 
> 
> 
> 但是直接运行起来还是报错，以下是转
>  请问你解决这个问题了吗，我也是同样的问题，用的cuda-12.1 tensorRT8.6.1.6版本

那你后来有运行成功吗，我运行的结果如下：[02/08/2025-19:27:41] [I] [TRT] ----------------------------------------------------------------
[02/08/2025-19:27:41] [I] [TRT] Input filename:   /home/kinggreat24/air_slam_ws/src/AirSLAM-master/output/superpoint_lightglue.onnx
[02/08/2025-19:27:41] [I] [TRT] ONNX IR version:  0.0.8
[02/08/2025-19:27:41] [I] [TRT] Opset version:    17
[02/08/2025-19:27:41] [I] [TRT] Producer name:    pytorch
[02/08/2025-19:27:41] [I] [TRT] Producer version: 2.1.0
[02/08/2025-19:27:41] [I] [TRT] Domain:
[02/08/2025-19:27:41] [I] [TRT] Model version:    0
[02/08/2025-19:27:41] [I] [TRT] Doc string:
[02/08/2025-19:27:41] [I] [TRT] ----------------------------------------------------------------
[02/08/2025-19:27:41] [W] [TRT] onnx2trt_utils.cpp:374: Your ONNX model has been generated with INT64 weights, while TensorRT does not natively support INT64. Attempting to cast down to INT32.
[02/08/2025-19:27:41] [W] [TRT] onnx2trt_utils.cpp:400: One or more weights outside the range of INT32 was clamped
[02/08/2025-19:27:41] [I] [TRT] Graph optimization time: 0.0851698 seconds.
[02/08/2025-19:27:41] [I] [TRT] Local timing cache in use. Profiling results in this builder pass will not be stored.
[02/08/2025-19:31:16] [I] [TRT] Detected 4 inputs and 1 output network tensors.
[02/08/2025-19:31:17] [I] [TRT] Total Host Persistent Memory: 15456
[02/08/2025-19:31:17] [I] [TRT] Total Device Persistent Memory: 0
[02/08/2025-19:31:17] [I] [TRT] Total Scratch Memory: 18874368
[02/08/2025-19:31:17] [I] [TRT] [MemUsageStats] Peak memory usage of TRT CPU/GPU memory allocators: CPU 21 MiB, GPU 171 MiB
[02/08/2025-19:31:17] [I] [TRT] [BlockAssignment] Started assigning block shifts. This will take 14 steps to complete.
[02/08/2025-19:31:17] [I] [TRT] [BlockAssignment] Algorithm ShiftNTopDown took 0.091921ms to assign 8 blocks to 14 nodes requiring 22282240 bytes.
[02/08/2025-19:31:17] [I] [TRT] Total Activation Memory: 22282240
[02/08/2025-19:31:17] [W] [TRT] TensorRT encountered issues when converting weights between types and that could affect accuracy.
[02/08/2025-19:31:17] [W] [TRT] If this is not the desired behavior, please modify the weights or retrain with regularization to adjust the magnitude of the weights.
[02/08/2025-19:31:17] [W] [TRT] Check verbose logs for the list of affected weights.
[02/08/2025-19:31:17] [W] [TRT] - 108 weights are affected by this issue: Detected subnormal FP16 values.
[02/08/2025-19:31:17] [W] [TRT] - 8 weights are affected by this issue: Detected values less than smallest positive FP16 subnormal value and converted them to the FP16 minimum subnormalized value.
[02/08/2025-19:31:17] [I] [TRT] [MemUsageChange] TensorRT-managed allocation in building engine: CPU +0, GPU +31, now: CPU 0, GPU 31 (MiB)
debug 5.1
[02/08/2025-19:31:17] [I] [TRT] Loaded engine size: 36 MiB
[02/08/2025-19:31:17] [I] [TRT] [MemUsageChange] TensorRT-managed allocation in engine deserialization: CPU +0, GPU +30, now: CPU 0, GPU 30 (MiB)
debug 6
debug 6.1
debug 7
[02/08/2025-19:31:17] [E] [TRT] 3: [runtime.cpp::~Runtime::346] Error Code 3: API Usage Error (Parameter check failed at: runtime/rt/runtime.cpp::~Runtime::346, condition: mEngineCounter.use_count() == 1. Destroying a runtime before destroying deserialized engines created by the runtime leads to undefined behavior.
)
map_builder done
dataset done
i ====== 0
[visual_odometry-2] process has died [pid 298841, exit code -11, cmd /home/kinggreat24/air_slam_ws/devel/lib/air_slam/visual_odometry __name:=visual_odometry __log:=/home/kinggreat24/.ros/log/b13e6ab4-e60f-11ef-b965-d358db09896b/visual_odometry-2.log].
log file: /home/kinggreat24/.ros/log/b13e6ab4-e60f-11ef-b965-d358db09896b/visual_odometry-2*.log

silence-moon, 2025-02-08:
没有。你可以试试使用docker运行，作者推荐

kinggreat24, 2025-02-08:
十分感谢，我在docker里面同样也有这个问题，哎，你试过docker了吗

kinggreat24, 2025-02-08:
> 没有。你可以试试使用docker运行，作者推荐
> […](#)
> 
> ---原始邮件---
> 发件人: ***@***.***&gt;
> 发送时间: 2025年2月8日(周六) 晚上7:34
> 收件人: ***@***.***&gt;;
> 抄送: ***@***.***&gt;;"State ***@***.***&gt;;
> 
> 
> 
> 
> 
> 我这边应该是启动是的tensorrt模型没有转成功的原因
>  …
> 
> ---原始邮件---
>  发件人: @.&gt;
>  发送时间: 2025年2月8日(周六) 晚上7:16
>  收件人: @.&gt;;
>  抄送: @.&gt;;"State @.&gt;;
> 
> 但是直接运行起来还是报错，以下是转
>  请问你解决这个问题了吗，我也是同样的问题，用的cuda-12.1 tensorRT8.6.1.6版本
> 
> 
> 那你后来有运行成功吗，我运行的结果如下：[02/08/2025-19:27:41] [I] [TRT] ----------------------------------------------------------------
>  [02/08/2025-19:27:41] [I] [TRT] Input filename:   /home/kinggreat24/air_slam_ws/src/AirSLAM-master/output/superpoint_lightglue.onnx
>  [02/08/2025-19:27:41] [I] [TRT] ONNX IR version:  0.0.8
>  [02/08/2025-19:27:41] [I] [TRT] Opset version:    17
>  [02/08/2025-19:27:41] [I] [TRT] Producer name:    pytorch
>  [02/08/2025-19:27:41] [I] [TRT] Producer version: 2.1.0
>  [02/08/2025-19:27:41] [I] [TRT] Domain:
>  [02/08/2025-19:27:41] [I] [TRT] Model version:    0
>  [02/08/2025-19:27:41] [I] [TRT] Doc string:
>  [02/08/2025-19:27:41] [I] [TRT] ----------------------------------------------------------------
>  [02/08/2025-19:27:41] [W] [TRT] onnx2trt_utils.cpp:374: Your ONNX model has been generated with INT64 weights, while TensorRT does not natively support INT64. Attempting to cast down to INT32.
>  [02/08/2025-19:27:41] [W] [TRT] onnx2trt_utils.cpp:400: One or more weights outside the range of INT32 was clamped
>  [02/08/2025-19:27:41] [I] [TRT] Graph optimization time: 0.0851698 seconds.
>  [02/08/2025-19:27:41] [I] [TRT] Local timing cache in use. Profiling results in this builder pass will not be stored.
>  [02/08/2025-19:31:16] [I] [TRT] Detected 4 inputs and 1 output network tensors.
>  [02/08/2025-19:31:17] [I] [TRT] Total Host Persistent Memory: 15456
>  [02/08/2025-19:31:17] [I] [TRT] Total Device Persistent Memory: 0
>  [02/08/2025-19:31:17] [I] [TRT] Total Scratch Memory: 18874368
>  [02/08/2025-19:31:17] [I] [TRT] [MemUsageStats] Peak memory usage of TRT CPU/GPU memory allocators: CPU 21 MiB, GPU 171 MiB
>  [02/08/2025-19:31:17] [I] [TRT] [BlockAssignment] Started assigning block shifts. This will take 14 steps to complete.
>  [02/08/2025-19:31:17] [I] [TRT] [BlockAssignment] Algorithm ShiftNTopDown took 0.091921ms to assign 8 blocks to 14 nodes requiring 22282240 bytes.
>  [02/08/2025-19:31:17] [I] [TRT] Total Activation Memory: 22282240
>  [02/08/2025-19:31:17] [W] [TRT] TensorRT encountered issues when converting weights between types and that could affect accuracy.
>  [02/08/2025-19:31:17] [W] [TRT] If this is not the desired behavior, please modify the weights or retrain with regularization to adjust the magnitude of the weights.
>  [02/08/2025-19:31:17] [W] [TRT] Check verbose logs for the list of affected weights.
>  [02/08/2025-19:31:17] [W] [TRT] - 108 weights are affected by this issue: Detected subnormal FP16 values.
>  [02/08/2025-19:31:17] [W] [TRT] - 8 weights are affected by this issue: Detected values less than smallest positive FP16 subnormal value and converted them to the FP16 minimum subnormalized value.
>  [02/08/2025-19:31:17] [I] [TRT] [MemUsageChange] TensorRT-managed allocation in building engine: CPU +0, GPU +31, now: CPU 0, GPU 31 (MiB)
>  debug 5.1
>  [02/08/2025-19:31:17] [I] [TRT] Loaded engine size: 36 MiB
>  [02/08/2025-19:31:17] [I] [TRT] [MemUsageChange] TensorRT-managed allocation in engine deserialization: CPU +0, GPU +30, now: CPU 0, GPU 30 (MiB)
>  debug 6
>  debug 6.1
>  debug 7
>  [02/08/2025-19:31:17] [E] [TRT] 3: [runtime.cpp::~Runtime::346] Error Code 3: API Usage Error (Parameter check failed at: runtime/rt/runtime.cpp::~Runtime::346, condition: mEngineCounter.use_count() == 1. Destroying a runtime before destroying deserialized engines created by the runtime leads to undefined behavior.
>  )
>  map_builder done
>  dataset done
>  i ====== 0
>  [visual_odometry-2] process has died [pid 298841, exit code -11, cmd /home/kinggreat24/air_slam_ws/devel/lib/air_slam/visual_odometry __name:=visual_odometry __log:=/home/kinggreat24/.ros/log/b13e6ab4-e60f-11ef-b965-d358db09896b/visual_odometry-2.log].
>  log file: /home/kinggreat24/.ros/log/b13e6ab4-e60f-11ef-b965-d358db09896b/visual_odometry-2*.log

十分感谢，我在docker里面同样也有这个问题，你试过docker了吗

silence-moon, 2025-02-08:
我没有试docker

silence-moon, 2025-02-08:
你可以把问题说的再详细一点，然后等作者看看回复吧

luohongk, 2025-04-16:
我用docker也是这个问题，我想问一下你解决了吗？