sair-lab/AirSLAM #128: AirVO and AirSLAM line feature problem

cpymaple, 2024-08-16:
Hi, thanks for the new work. I have two questions about the AirSLAM/AirVO:

1. AirVO/AirSLAM both use the point and line features. I wonder know whether you test the line feature ablation experiments in AirVO/AirSLAM framework  (The framework "only use points features" and "points with lines features")? 
2. AirSLAM adds the lightglue.onnx. Is it possible to release the "connver2onnx" folder like AirVO in the future?

Thanks for your time~

xukuanHIT, 2024-08-16:
@cpymaple Hi,
1. In AirVO, we have compared the system with and without line features. The results can be found in the IROS paper. For AirSLAM, we focus more on map-based localization, so we give the relocalization recall results with and without the use of line features in the ablation study.
2. Sure. It will be released after the paper is accepted.
3. No, the feature detection network, PLNet, relies on some features of TensorRT 8.6. But I am considering to create a point-only system to reduce dependency on specific GPUs and development environments

cpymaple, 2024-08-16:
> @cpymaple Hi,
>
>     1. In AirVO, we have compared the system with and without line features. The results can be found in the IROS paper. For AirSLAM, we focus more on map-based localization, so we [...]

Thanks for your quick reply~

closed 2024-08-16 by cpymaple