sair-lab/AirSLAM #130: 关于运行自己数据集的一点问题

jingyang-huang, 2024-08-17:
作者您好！最近关注到这篇工作，希望能借鉴一些方法来解决我们数据集中的问题，但是我们发现在运行时遇到了两个问题，一个是运行vo时里程计出现了较大的漂移，如图1、2所示，
[image]
[image]

我们使用同样的内外参和数据包，对比了vins fusion和air slam的轨迹，发现两者都存在漂移情况，但是似乎air slam的漂移更大一点
还有一个是回环时，程序报错
```
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.2.0) ../modules/imgproc/src/imgwarp.cpp:668: error: (-215:Assertion failed) !ssize.empty() in function 'remapBilinear'
``` 
我们已经排查了参数image_width和image_height的大小，这两个应该是正确的，想请问下有什么办法可以让它正常运行

xukuanHIT, 2024-08-18:
@jingyang-huang 这种feature跟踪正常，但是轨迹错误的情况一般都是畸变参数和外参的问题。不知道你有没有用IMU，可以先用纯双目试试。对于双目校正后的双目图像，你可以把左右双目图像并排放，每个若干像素画一条绿线，检查下左右双目对应的点是否在同一高度上，以此来检查下畸变和外参有没有问题

jingyang-huang, 2024-08-19:
> 我们怀疑可能是输入外参的问题，对照作者提供的realsense_848_480.yaml进行了修改，内参都由realsense的camera_info提供，且是纯vo，发现前半程运行正常，后半程出现了较大的漂移

作者您好，我们也感觉可能是输入内外参的问题，我们对照提供的realsense_848_480.yaml进行了修改，内参都由realsense的camera_info提供，外参保持一致，且是纯vo，发现前半程运行正常，后半程出现了较大的漂移
```
%YAML:1.0

image_height: 480
image_width: 640
use_imu: 0

depth_lower_thr: 0.1
depth_upper_thr: 10.0
max_y_diff: 2

# Calibration
distortion_type: 0  # 0 for undistorted inputs, 1 for radial-tangential: [k1, k2, p1, p2, k3], 2 for equidistant/fisheye:  [k1, k2, k3, k4, 0].
cam0:
  intrinsics: [385.6853332519531,385.6853332519531, 323.28155517578125, 232.31861877441406] # fx, fy, cx, cy
  distortion_coeffs: [0.0, 0.0, 0.0, 0.0, 0]
  T_type: 0           # 0 for Euroc format, the following T is Tbc. 1 for Kalibr format, the following T is Tcb
  T: 
  - [1.0, 0.0, 0.0, 0.0]
  - [0.0, 1.0, 0.0, 0.0]
  - [0.0, 0.0, 1.0, 0.0]
  - [0.0, 0.0, 0.0, 1.0]
cam1:
  intrinsics: [385.6853332519531,385.6853332519531, 323.28155517578125, 232.31861877441406] # fx, fy, cx, cy
  distortion_coeffs: [0.0, 0.0, 0.0, 0.0, 0]
  T_type: 0           
  T: 
  - [1.0, 0.0, 0.0, 0.05]
  - [0.0, 1.0, 0.0, 0.0]
  - [0.0, 0.0, 1.0, 0.0]
  - [0.0, 0.0, 0.0, 1.0]

# IMU
rate_hz: 200
gyroscope_noise_density: 0.05    # [ rad / s / sqrt(Hz) ]   ( gyro "white noise" )
gyroscope_random_walk: 0.0005       # [ rad / s^2 / sqrt(Hz) ] ( gyro bias diffusion )
accelerometer_noise_density: 0.5  # [ m / s^2 / sqrt(Hz) ]   ( accel "white noise" )
accelerometer_random_walk: 0.005    # [ m / s^3 / sqrt(Hz) ].  ( accel bias diffusion )
g_value: 9.805
```

xukuanHIT, 2024-08-19:
@jingyang-huang camera_info提供的不会很准，你可以用我上面说的方法验证下双目是否对齐。VINS会把一些相机参数加到优化里面，AirSLAM用的是固定相机参数，所以对参数更敏感一些

jingyang-huang, 2024-08-21:
> @jingyang-huang camera_info提供的不会很准，你可以用我上面说的方法验证下双目是否对齐。VINS会把一些相机参数加到优化里面，AirSLAM用的是固定相机参数，所以对参数更敏感一些

好的，可能确实是一些参数的影响，我们会再验证下双目是否对齐，不过目前我们的试验没有开VINS的外参优化

closed 2024-08-21 by jingyang-huang