sair-lab/AirSLAM #134: Code is wrong!!!!!!!!

liwenlon, 2024-08-20:
Please check your "  edge_project_line.cc ".  There is a clear error in line 39.  This should be " line_2d(0) = fx * w(0)".

xukuanHIT, 2024-08-20:
@liwenlon Thank you for pointing out the mistake; I've corrected it. But in fact, this error won't impact the performance, because after stereo rectification, fx = fy.

liwenlon, 2024-08-21:
Yes, i guess there will be equal parameters in the system. I have another question. I found that you write " FrameOptimization“ in the " MapBuilder  ".   However, you defined the line constraint, as "    MapOfLine3d lines;  ". But I noticed that you haven't assigned any line constraints. Did your line constraint work in the FrameOptimization function.   Please look line 323 and line 393 in "map_builder .cc  "

xukuanHIT, 2024-08-21:
Answered in #136

closed 2024-08-21 by xukuanHIT