sair-lab/AirSLAM #140: AirSLAM config parameter setting and superpoint image resize

cpymaple, 2024-08-22:
Hi, I use AirSLAM to run my own dataset without imu (camera resolution are 346 * 260). I follow the euroc config to revise parameter. However, the terminal shows the "good_stereo_points = 20 - 30" so that the systems cannot complete initialization. I would like to ask should I need to change the parameters in "vo_mydata.yaml"? To be specifically, should I need to revise "plnet, keyframe, and optimization" parameter? Are there some tricks to tuning the parameters in this part? 

By the way, I notice that the superpoint in AirSLAM add the "image resize (512 * 512)". Is this so that the superpoint method can be used on different camera resolutions? If I use a camera with a lower resolution than this, will this have an effect on feature point extraction or lightglue/superglue matching?

Thanks for your time~

xukuanHIT, 2024-08-23:
@cpymaple Hi,
1. You can try lowering the keypoint_threshold to obtain more keypoints.
2. Due to TensorRT not supporting some of PyTorch's operators, the current PLNet can only be resized to 512x512. We are waiting for the update of TensorRT to solve this issue. Besides, we are also considering creating a point-only version to be more widely compatible with different hardware and image inputs.

cpymaple, 2024-08-24:
> @cpymaple Hi,
>
>     1. You can try lowering the keypoint_threshold to obtain more keypoints.
>
>     2. Due to TensorRT not supporting some of PyTorch's operators, the current PLNet can only be resized [...]

Thanks for your quick reply~

closed 2024-08-24 by cpymaple