sair-lab/AirSLAM #142: development environment

dapao-0808, 2024-08-30:
Hi, thank you very much for providing the paper! May I ask what is your development environment on Nvidia Jetson Orin, and what are the versions of JetPack, CUDA, CUDNN, and TensorRT? Is there an ARM based Docker?

xukuanHIT, 2024-08-31:
@dapao-0808 Hi, we use JetPack 6.0, and the default  CUDA, CUDNN, and TensorRT in JetPack 6.0.

dapao-0808, 2024-09-02:
Thank you, the JatPack version I am currently using is still 5.1.3

dapao-0808, 2024-09-09:
@xukuanHIT Hi, I have upgraded JetPack to 6.0, but the ROS version has also been changed to ROS2. I wonder if it is possible to open source the code and launch files related to ROS2. Thank you very much!

dapao-0808, 2024-09-09:
Well, thank you so much.