sair-lab/AirSLAM #143: Embedded Operation

zw-92, 2024-09-05:
Hello, I am planning to run this method on the RK3576 board. I would like to know if the 6 TOPS of computing power is sufficient (the actual computing power is 1200 GFLOPS), and what the frame rate can reach. What is the computing power of the embedded device you are using, and could you please inform me of the model? I look forward to your reply.

xukuanHIT, 2024-09-06:
@zw-92 Hi, we know little about your platform. The test results in the paper for embedded platforms were conducted on the Jetson Orin. The detailed information about the Jetson Orin is available on its official website.

zw-92, 2024-09-20:
Hi，抱歉打扰您了。我在配置文件中没有看到双目/立体相机有关“基线”的配置，是使用了默认的吗？

zw-92, 2024-09-23:
Hi，请问下PL-Net的网络代码可以开源吗？期待您的回复。

xukuanHIT, 2024-09-23:
PLNet的训练代码会在论文接收后开源

closed 2024-10-28 by xukuanHIT