sair-lab/AirSLAM #145: 无法在Newer College Dataset上运行

Alex-Beh, 2024-09-10:
你好 我尝试在[Newer College Dataset](https://ori-drs.github.io/newer-college-dataset/)上测试AirSLAM，可是一直没有[good_stereo_point](https://github.com/sair-lab/AirSLAM/blob/61779f24d104baaf9f447cf8a5f885cc07970118/src/map_builder.cc#L123). 有matches可是应该都被[这部分的checking](https://github.com/sair-lab/AirSLAM/blob/61779f24d104baaf9f447cf8a5f885cc07970118/src/frame.cc#L154-L156)给filter掉了。请问应该如何设置这里的parameter?
https://github.com/sair-lab/AirSLAM/blob/61779f24d104baaf9f447cf8a5f885cc07970118/configs/camera/euroc.yaml#L7-L9

[vo_oxford.json](https://github.com/user-attachments/files/16938320/vo_oxford.json)
[oxford.json](https://github.com/user-attachments/files/16938321/oxford.json)

[需要rename去yaml的extension，github不允许上传yaml文件]

Alex-Beh, 2024-09-10:
了解！请问你有在Hilti和NTU VIRAL DATASET上运行的config吗？

xukuanHIT, 2024-09-10:
没有在这两个数据集上跑过

closed 2024-10-10 by xukuanHIT