sair-lab/AirSLAM #146: 怎么在Jetson Orin NX上部署

jingyang-huang, 2024-09-10:
NX只有jetpack 6.0 支持 Cuda 12.x以及Tensorrt 8.6，但是Jetpack 6.0上没法安装ros noetic，请问是怎么在Jetson Orin NX上部署的呀

xukuanHIT, 2024-09-10:
@jingyang-huang 你好，我们是源码安装的ROS1。由于PLNet转TensorRT依赖Tensorrt 8.6，我们也在开发一个纯点的系统以降低对Cuda和Tensorrt版本的要求

niceday-hh, 2024-09-10:
@xukuanHIT 你好，我想问一下jetpack5.1版本只能安装tensorrt8.5，能不能兼容tensorrt8.6的代码呢？我编译了发现有诸如error: 'tempfilecontrolflags' in namespace 'nvinfer1' deos not name a type 这类报错

jingyang-huang, 2024-09-19:
> @jingyang-huang 你好，我们是源码安装的ROS1。由于PLNet转TensorRT依赖Tensorrt 8.6，我们也在开发一个纯点的系统以降低对Cuda和Tensorrt版本的要求

感谢您的回复，我们发现lightglue依赖tensorrt8.6,部署比较困难，所以最后选择的superglue，它能够在tensorrt8.5上量化

xukuanHIT, 2024-09-19:
> @xukuanHIT 你好，我想问一下jetpack5.1版本只能安装tensorrt8.5，能不能兼容tensorrt8.6的代码呢？我编译了发现有诸如error: 'tempfilecontrolflags' in namespace 'nvinfer1' deos not name a type 这类报错

暂时不能

niceday-hh, 2024-09-19:
> > @xukuanHIT 你好，我想问一下jetpack5.1版本只能安装tensorrt8.5，能不能兼容tensorrt8.6的代码呢？我编译了发现有诸如error: 'tempfilecontrolflags' in namespace 'nvinfer1' deos not name a type 这类报错
> 
> 暂时不能

好的，感谢您的回复！

closed 2024-09-24 by jingyang-huang