sair-lab/AirSLAM #147: why

chuichuiY, 2024-09-14:
[image]

xukuanHIT, 2024-09-18:
@chuichuiY Hi, have you installed OpenCV?

chuichuiY, 2024-09-18:
> @chuichuiY你好，你安装了OpenCV吗？

yes,I downloaded opencv4.2.0 and this problem occurred when compiling AIRSLAM.

xukuanHIT, 2024-09-18:
OpenCV is not found when compiling DBoW2. Can you check your installation of the OpenCV? For example, can you find the path of OpenCV ".hpp" and ".so" files on your computer?

chuichuiY, 2024-09-18:
[image]

xukuanHIT, 2024-09-18:
They are building results. Have you installed them?

xukuanHIT, 2024-09-18:
@chuichuiY https://docs.opencv.org/4.x/d7/d9f/tutorial_linux_install.html

chuichuiY, 2024-09-18:
[image]

> They are building results. Have you installed them?

xukuanHIT, 2024-09-18:
Can you find opencv2/opencv.hpp?

chuichuiY, 2024-09-18:
> Can you find opencv2/opencv.hpp?

yes ,@wp:~/opencv-4.2.0/include/opencv2/opencv.hpp

xukuanHIT, 2024-09-18:
OK, can you find the OpenCVConfig*.cmake?

chuichuiY, 2024-09-18:
> OK, can you find the OpenCVConfig*.cmake in /usr/local/cmake/opencv4?

/usr/local/stow/absl/lib/cmake/opencv4/OpenCVConfig-version.cmake
/usr/local/stow/absl/lib/cmake/opencv4/OpenCVConfig.cmake

xukuanHIT, 2024-09-18:
I am not sure why the Cmakelist can not find your opencv. You can set the path in the Cmakelist using "set(OpenCV_DIR /usr/local/stow/absl/lib/cmake/opencv4)".  You can also create a small project to test whether your opencv can be linked correctly.

chuichuiY, 2024-09-18:
> I am not sure why the Cmakelist can not find your opencv. You can set the path in the Cmakelist using "set(OpenCV_DIR /usr/local/stow/absl/lib/cmake/opencv4)". You can also create a small project to test whether your opencv can be linked correctly.

Thank you so much. It is true that DBOW2 cannot find my opencv, and it can be compiled successfully after modification. I would like to ask what data sets this can run on, and do I need to use rosbag to run it?

chuichuiY, 2024-09-18:
[image]

xukuanHIT, 2024-09-18:
Please use the original dataset. We use  the image name to get the time stamp

chuichuiY, 2024-09-19:
> ![1726675087596](https://private-user-images.githubusercontent.com/111644261/368645428-590c8377-8520-4eda-8316-6b26ed2f61c8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY3MzAyMzcsIm5iZiI6MTcyNjcyOTkzNywicGF0aCI6Ii8xMTE2NDQyNjEvMzY4NjQ1NDI4LTU5MGM4Mzc3LTg1MjAtNGVkYS04MzE2LTZiMjZlZDJmNjFjOC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTE5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkxOVQwNzEyMTdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jMTVhMzlhOTZmYzkyNTVkMzJlYjZhYWRlYWJmZmUxMGM2MjI4NjQxNjdmNDlkNmM4NDM0NTI0NmNkYTM3YzdkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.E4ChldczfGFpPvwl85zRKwxl92suC59izhEoPDmQShY)
Is this error due to the data set? I changed the name of the MH01 data and the .launch file as required.

xukuanHIT, 2024-09-19:
Please use the original dataset. Don't change the image name.

closed 2025-01-22 by xukuanHIT