sair-lab/AirSLAM #148: Fill in the external parameters for binocular purpose

WangHongxuan0715, 2024-09-19:
If I use my own recorded data to fill in the external parameters, how can I modify it based on the oivio.yaml configuration file? Currently, I fill in the transformation T from the left camera to the right camera in cam0, and then fill in the unit matrix in cam1. This filling shows that almost no matching feature points are detected. I think I may have filled it in incorrectly.

%YAML:1.0

image_height: 1260
image_width: 1680
use_imu: 0

depth_lower_thr: 0.1
depth_upper_thr: 10.0
max_y_diff: 30

# Calibration
distortion_type: 1  # 0 for undistorted inputs, 1 for radial-tangential: [k1, k2, p1, p2, k3], 2 for equidistant/fisheye:  [k1, k2, k3, k4, 0].
cam0:
  intrinsics: [1192.703518726793618, 1192.437900413209945, 839.587888923699097, 609.982668736013238] # fx, fy, cx, cy
  distortion_coeffs: [0, 0, 0, 0, 0]
  T_type: 0           # 0 for Euroc format, the following T is Tbc. 1 for Kalibr format, the following T is Tcb
  T: 
  - [0.999999606792653, 0.000789311607587, 0.000404229793522, -0.430370755417723160]
  - [-0.000772038232196, 0.999155430938691, -0.041083193433401, -0.002471955916083025]
  - [-0.000436315834999, 0.041082865198333, 0.999155647442173, -0.015380798799766124]
  - [0.0, 0.0, 0.0, 1.0]
cam1:
  intrinsics: [1174.011910797035171, 1175.708934480511516, 819.071816376535480, 609.982668736013238] 
  distortion_coeffs: [0,  0, 0, 0, 0]
  T_type: 0           
  T: 
  - [1, 0, 0, 0]
  - [0, 1, 0, 0]
  - [0, 0, 1, 0]
  - [0.0, 0.0, 0.0, 1.0]

Anju023, 2024-09-19:
I think this will help you.

https://mega.nz/file/1vkGlJTL#yU-SJLlIilF0mx5qwpPOd3BM8iKZqOJLEQUStR76rrQ
If you don't have the c compliator, install it.(gcc or clang)

xukuanHIT, 2024-09-19:
@WangHongxuan0715 Hi, [tartanair.yaml](https://github.com/sair-lab/AirSLAM/blob/master/configs/camera/tartanair.yaml) is a configuration file for stereo camera.

closed 2025-01-10 by xukuanHIT