sair-lab/AirSLAM #149: Error in FeatureDetector building

2080524785, 2024-09-25:
I used the docker pull environment to run Airslam, running on the Euroc dataset, but rizv didn't show the picture on it, and the process terminated after waiting for a while to only show a few frames of the runtime. 
Because I want to see the effect, I run on my own computer, and the graphics card is Nvidia RTX 1650.
I would like to ask if this question is related to my low graphics card, or if there is something else.

> command : `roslaunch air_slam vo_euroc.launch`

*  `bash output`
```
... logging to /root/.ros/log/4b4d5338-7b24-11ef-95b3-0242ac110002/roslaunch-e5eda9553952-12462.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://e5eda9553952:34165/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /visual_odometry/camera_config_path: /workspace/catkin...
 * /visual_odometry/config_path: /workspace/catkin...
 * /visual_odometry/dataroot: /workspace/datase...
 * /visual_odometry/model_dir: /workspace/catkin...
 * /visual_odometry/saving_dir: /workspace/catkin...

NODES
  /
    rviz (rviz/rviz)
    visual_odometry (air_slam/visual_odometry)

ROS_MASTER_URI=http://localhost:11311

process[visual_odometry-1]: started with pid [12487]
process[rviz-2]: started with pid [12488]
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
QStandardPaths: wrong ownership on runtime directory /tmp/runtime-root, 1000 instead of 0
[ INFO] [1727263562.583094810]: rviz version 1.14.20
[ INFO] [1727263562.583129521]: compiled against Qt version 5.12.8
[ INFO] [1727263562.583140695]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1727263562.595139275]: Forcing OpenGl version 0.
config_file = /workspace/catkin_ws/src/AirSLAM/configs/visual_odometry/vo_euroc.yaml
config done
[ INFO] [1727263562.837373237]: Stereo is NOT SUPPORTED
[ INFO] [1727263562.837479186]: OpenGL device: llvmpipe (LLVM 12.0.0, 256 bits)
[ INFO] [1727263562.837523815]: OpenGl version: 3.1 (GLSL 1.4).
Error in FeatureDetector building
map_builder done
dataset done
i ====== 0
One Frame Processinh Time: 2 ms.
i ====== 1
One Frame Processinh Time: 2 ms.
i ====== 2
One Frame Processinh Time: 2 ms.
i ====== 3
One Frame Processinh Time: 3 ms.
i ====== 4
One Frame Processinh Time: 1 ms.
i ====== 5
[visual_odometry-1] process has died [pid 12487, exit code -11, cmd /workspace/catkin_ws/devel/lib/air_slam/visual_odometry __name:=visual_odometry __log:=/root/.ros/log/4b4d5338-7b24-11ef-95b3-0242ac110002/visual_odometry-1.log].
log file: /root/.ros/log/4b4d5338-7b24-11ef-95b3-0242ac110002/visual_odometry-1*.log

```
* `log file`
```
[roslaunch][INFO] 2024-09-25 19:17:36,458: Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
[roslaunch][INFO] 2024-09-25 19:17:36,469: Done checking log file disk usage. Usage is <1GB.
[... 59 lines cut ...]
[roslaunch][ERROR] 2024-09-25 19:19:57,251: [visual_odometry-1] process has died [pid 11399, exit code -11, cmd /workspace/catkin_ws/devel/lib/air_slam/visual_odometry __name:=visual_odometry __log:=/root/.ros/log/4b4d5338-7b24-11ef-95b3-0242ac110002/visual_odometry-1.log].
log file: /root/.ros/log/4b4d5338-7b24-11ef-95b3-0242ac110002/visual_odometry-1*.log
[roslaunch.pmon][INFO] 2024-09-25 19:19:57,251: ProcessMonitor.unregister[visual_odometry-1] starting
[roslaunch.pmon][INFO] 2024-09-25 19:19:57,251: ProcessMonitor.unregister[visual_odometry-1] complete
[roslaunch.pmon][INFO] 2024-09-25 19:20:31,585: ProcessMonitor.shutdown <ProcessMonitor(ProcessMonitor-1, started daemon 140314676979456)>
[roslaunch.pmon][INFO] 2024-09-25 19:20:31,638: ProcessMonitor._post_run <ProcessMonitor(ProcessMonitor-1, started daemon 140314676979456)>
[roslaunch.pmon][INFO] 2024-09-25 19:20:31,639: ProcessMonitor._post_run <ProcessMonitor(ProcessMonitor-1, started daemon 140314676979456)>: remaining procs are [<roslaunch.nodeprocess.LocalProcess object at 0x7f9da88ecac0>]
[roslaunch.pmon][INFO] 2024-09-25 19:20:31,639: ProcessMonitor exit: killing rviz-2
[roslaunch][INFO] 2024-09-25 19:20:31,639: [rviz-2] killing on exit
[roslaunch][INFO] 2024-09-25 19:20:31,640: process[rviz-2]: killing os process with pid[11400] pgid[11400]
[roslaunch][INFO] 2024-09-25 19:20:31,640: [rviz-2] sending SIGINT to pgid [11400]
[roslaunch][INFO] 2024-09-25 19:20:31,640: [rviz-2] sent SIGINT to pgid [11400]
[roslaunch][INFO] 2024-09-25 19:20:33,844: process[rviz-2]: SIGINT killed with return value 0
[roslaunch.pmon][INFO] 2024-09-25 19:20:33,845: ProcessMonitor exit: cleaning up data structures and signals
[roslaunch.pmon][INFO] 2024-09-25 19:20:33,845: ProcessMonitor exit: pmon has shutdown
[roslaunch][INFO] 2024-09-25 19:20:33,845: process monitor is done spinning, initiating full shutdown
[roslaunch][INFO] 2024-09-25 19:20:33,845: runner.stop()
[roslaunch][INFO] 2024-09-25 19:20:33,845: shutting down processing monitor...
[roslaunch][INFO] 2024-09-25 19:20:33,845: shutting down processing monitor <ProcessMonitor(ProcessMonitor-1, stopped daemon 140314676979456)>
[roslaunch.pmon][INFO] 2024-09-25 19:20:33,845: ProcessMonitor.shutdown <ProcessMonitor(ProcessMonitor-1, stopped daemon 140314676979456)>
[roslaunch][INFO] 2024-09-25 19:20:33,845: ... shutting down processing monitor complete
[roslaunch][INFO] 2024-09-25 19:20:33,845: done
[roslaunch.pmon][INFO] 2024-09-25 19:20:33,846: ProcessMonitor.shutdown <ProcessMonitor(ProcessMonitor-1, stopped daemon 140314676979456)>
[rospy.core][INFO] 2024-09-25 19:20:33,846: signal_shutdown [atexit]
```

2080524785, 2024-09-25:
Sorry, I don't why it It works fine now.

xukuanHIT, 2024-09-25:
@2080524785 Hi, currently, in the Docker environment, we are using CUDA 12.1 and TensorRT 8.6. I think it’s possible that your NVIDIA driver and GPU don’t support these versions. We are also creating a point-only system to support a wider range of hardware

2080524785, 2024-09-25:
Thank you for your reply, this should be my personal computer reason, I was able to work fine on the euroc dataset after multiple attempts. Also, I'd like to ask if this checkpoint-only part needs to be modified or if it hasn't been released yet.

2080524785, 2024-09-25:
> @2080524785 Hi, currently, in the Docker environment, we are using CUDA 12.1 and TensorRT 8.6. I think it’s possible that your NVIDIA driver and GPU don’t support these versions. We are also creating a point-only system to support a wider range of hardware

Thank you for your reply, this should be my personal computer reason, I was able to work fine on the euroc dataset after multiple attempts. Also, I'd like to ask if this checkpoint-only part needs to be modified or if it hasn't been released yet.

2080524785, 2024-09-26:
> It hasn't been released yet. Currently, the C++ code of PLNet relies on TensorRT 8.6 and Cuda 12.1, so in the point-only system, we will replace it with superpoint.

Thanks, I look forward to your work.

closed 2024-09-26 by 2080524785