sair-lab/AirSLAM #153: AirVO 运行 roslaunch airvo euroc.launch

2805651606, 2024-10-11:
i ===== 3679
One Frame Processinh Time: 34 ms.
i ===== 3680
One Frame Processinh Time: 31 ms.
i ===== 3681
One Frame Processinh Time: 34 ms.
Save file to /root/catkin_ws_vo/src/AirVO/debug/traj.txt
_keyframe_ids.size = 0
terminate called without an active exception
[air_vo-2] process has died [pid 237231, exit code -6, cmd /root/catkin_ws_vo/devel/lib/air_vo/air_vo __name:=air_vo __log:=/root/.ros/log/790f7780-87a2-11ef-a093-0242ac110003/air_vo-2.log].
log file: /root/.ros/log/790f7780-87a2-11ef-a093-0242ac110003/air_vo-2*.log
这是怎么回事呢

xukuanHIT, 2024-10-14:
@2805651606 你好，rviz里有特征点显式吗？

rangingwaves, 2024-10-15:
@xukuanHIT 你好，我运行 roslaunch airvo euroc.launch 也遇到了一个问题，就是rviz里面只显示点特征跟踪的照片，不显示地图点和轨迹。然后我发现应该是superglue的问题，我打印superglue的matches匹配对，发现全是0。请问我该怎么解决？

xukuanHIT, 2024-10-15:
@rangingwaves 你好，有生成superglue的engine文件吗？

rangingwaves, 2024-10-15:
@xukuanHIT 有的

xukuanHIT, 2024-10-15:
@rangingwaves 你的测试环境是怎么样的？

xukuanHIT, 2024-10-15:
@rangingwaves airvo需要tensorrt8.5。 两个tensorrt版本不太兼容

rangingwaves, 2024-10-15:
@xukuanHIT 好的·谢谢

2805651606, 2024-10-15:
> @2805651606 你好，rviz里有特征点显示吗？

有的，已经解决了

2805651606, 2024-10-15:
@xukuanHIT  你好，请问OiVIO数据集你有？目前github上的已经失效了

xukuanHIT, 2024-10-15:
@2805651606 UMA和OIVIO的数据集下载，可以参考这个 #70

DeepestLover3685, 2024-11-02:
@xukuanHIT 请问具体怎么设置dataroot
<arg name="dataroot" default = "/home/llk/catkin_ws/data/mav0/MH_01_easy.bat" />
这样可以么

DeepestLover3685, 2024-11-02:
@xukuanHIT  请问 具体怎么设置dataroot？<arg name="dataroot" default = "/home/llk/catkin_ws/data/mav0/MH_01_easy.bat" />
这样可以么

DeepestLover3685, 2024-11-02:
@xukuanHIT <arg name="dataroot" default = "/home/llk/catkin_ws/data/mav0/MH_01_easy.bat" />

DeepestLover3685, 2024-11-02:
/home/llk/catkin_ws/data/mav0/MH_01_easy.bat

closed 2024-12-17 by xukuanHIT