sair-lab/AirSLAM #155: 实时性

tomliuli, 2024-10-30:
请问一下，这个开源的代码能实时吗，感觉只能跑数据集

xukuanHIT, 2024-10-31:
可以将[demo](https://github.com/sair-lab/AirSLAM/tree/master/demo)下的几个例子改成ros接口

tomliuli, 2024-11-16:
我现在在NX使用的cuda12.2以及Tensorrt 8.6，出现下面的现象

[image]

tomliuli, 2024-11-16:
然后我根据issue，把(https://github.com/sair-lab/AirSLAM/blob/2a23e669f20da45824df486c6b326d84f4fbfb67/src/light_glue.cpp#L17C43-L17C58)改成 setReportableSeverity(Logger::Severity::kINFO); 输出更多些信息，是
yy@ubuntu:~/airslam$ roslaunch air_slam vo_euroc.launch

... logging to /home/yy/.ros/log/04c2b888-a41c-11ef-8eef-e8fb1c354549/roslaunch-ubuntu-14984.log

Checking log directory for disk usage. This may take a while.

Press Ctrl-C to interrupt

Done checking log file disk usage. Usage is <1GB.

started roslaunch server [http://ubuntu:39237/](http://ubuntu:39237/)

# SUMMARY

PARAMETERS

- /rosdistro: noetic

- /rosversion: 1.17.0

- /visual_odometry/camera_config_path: /home/yy/airslam/...

- /visual_odometry/config_path: /home/yy/airslam/...

- /visual_odometry/dataroot: /home/yy/MH_01_ea...

- /visual_odometry/model_dir: /home/yy/airslam/...

- /visual_odometry/saving_dir: /home/yy/airslam/...

NODES

/

rviz (rviz/rviz)

visual_odometry (air_slam/visual_odometry)

auto-starting new master
process[master]: started with pid [14993]
ROS_MASTER_URI=[http://localhost:11311/](http://localhost:11311)

setting /run_id to 04c2b888-a41c-11ef-8eef-e8fb1c354549
process[rosout-1]: started with pid [15003]
started core service [/rosout]
process[visual_odometry-2]: started with pid [15007]
process[rviz-3]: started with pid [15011]
config_file = /home/yy/airslam/src/AirSLAM/configs/visual_odometry/vo_euroc.yaml
config done
[11/16/2024-21:09:32] [W] [TRT] Unable to determine GPU memory usage
[11/16/2024-21:09:32] [W] [TRT] Unable to determine GPU memory usage
[11/16/2024-21:09:32] [I] [TRT] [MemUsageChange] Init CUDA: CPU +0, GPU +0, now: CPU 18, GPU 0 (MiB)
[11/16/2024-21:09:32] [W] [TRT] CUDA initialization failure with error: 35. Please check your CUDA installation:  [http://docs.nvidia.com/cuda/cuda-installation-guide-linux/index.html](http://docs.nvidia.com/cuda/cuda-installation-guide-linux/index.html)
Erron lightglue building
Error in SuperPoint building
[ INFO] [1731762573.101821101]: rviz version 1.14.25
[ INFO] [1731762573.101931987]: compiled against Qt version 5.12.8
[ INFO] [1731762573.101980181]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1731762573.114261325]: Forcing OpenGl version 0.
[visual_odometry-2] process has finished cleanly
log file: /home/yy/.ros/log/04c2b888-a41c-11ef-8eef-e8fb1c354549/visual_odometry-2*.log
[ INFO] [1731762574.115051890]: Stereo is NOT SUPPORTED
[ INFO] [1731762574.115219419]: OpenGL device: NVIDIA Tegra Orin (nvgpu)/integrated
[ INFO] [1731762574.115285726]: OpenGl version: 4.6 (GLSL 4.6).

tomliuli, 2024-11-16:
我在自己的pc上docker是能跑通的，但是在NX部署时出现了上面的报错

xukuanHIT, 2024-11-18:
@tomliuli 你好，这个问题貌似挺多人遇到了，每个人原因似乎不太一样，不过都是一些和TensorRT和Cuda相关的问题。你这个可能也是关于这两个的版本或者链接的环境问题，但具体是怎么回事我也不太确定

tomliuli, 2024-11-19:
接下来我准备重刷NX，请问一下，您们NX选的jetpack版本是多少（因为我这个系统的CUDA 12.2，TensorRT 8.6.1我是重新配置的，不是刷机的时候配置得，我准备这次直接刷机配置），还有opencv版本能选择opencv4.5.0吗

closed 2024-11-21 by tomliuli