sair-lab/AirSLAM #159: “No Image”

Lmy971109, 2024-10-31:
Dear author, thank you for your great contribution. However, after compiling, I executed "roslaunch air_slam vo_euroc.launch" but there was no data. The running results are as follows. What is the reason? I also used the data of EUROC V1_02_medium.
And the settings in vo_euroc.launch are as follows: 
[image]

[image]

closed 2024-10-31 by Lmy971109

zhanghanduo, 2024-11-23:
@Lmy971109 Hi. Did you resolve this issue? I also met the same problem without TensorRT running

WuQi-Dancer, 2025-02-05:
> Dear author, thank you for your great contribution. However, after compiling, I executed "roslaunch air_slam vo_euroc.launch" but there was no data. The running results are as follows. What is the [...]

Have you solved this problem?

DevRookiee, 2025-03-11:
The same question I want to ask!!!

luohongk, 2025-04-16:
The same question I want to ask!!!