sair-lab/AirSLAM #162: roslaunch air_slam vo_euroc.launch遇到了问题

DeepestLover3685, 2024-11-02:
@xukuanHIT 您好！我在roslaunch air_slam vo_euroc.launch时遇到了错误。
首先是rviz没有显示
[image]
终端有这样的报错
Erron lightglue building
Error in SuperPoint building

这是我vo_euroc.launch的配置以及dataroot路径
[image]

xukuanHIT, 2024-11-02:
@DeepestLover3685 你好，应该是环境没配好，TensorRT在转模型时没成功

DeepestLover3685, 2024-11-02:
> @DeepestLover3685 你好，应该是环境没配好，TensorRT在转模型时没成功

感谢，我的环境是 ubuntu20.04 tensorrt8.61 cuda12.1 我catkin_make AirSLAM时并没有问题

xukuanHIT, 2024-11-02:
@DeepestLover3685 你好，这个环境可以跑AirSLAM，但AirVO跑着有问题，TensorRT8.6和之前版本不兼容。AirVO具体环境要求可以参考它的[ReadME](https://github.com/sair-lab/AirSLAM/tree/airvo_iros)

xukuanHIT, 2024-11-02:
可以把[这一行](https://github.com/sair-lab/AirSLAM/blob/2a23e669f20da45824df486c6b326d84f4fbfb67/src/light_glue.cpp#L17C43-L17C58)改成  setReportableSeverity(Logger::Severity::kINFO); 输出更多些信息试试

DeepestLover3685, 2024-11-02:
> 可以把[这一行](https://github.com/sair-lab/AirSLAM/blob/2a23e669f20da45824df486c6b326d84f4fbfb67/src/light_glue.cpp#L17C43-L17C58)改成 setReportableSeverity（Logger：：Severity：：kINFO）;输出更多些信息试试

[11/02/2024-16:08:37] [I] [TRT] [MemUsageChange] Init CUDA: CPU +14, GPU +0, now: CPU 35, GPU 67 (MiB)
[11/02/2024-16:08:38] [I] [TRT] [MemUsageChange] Init builder kernel library: CPU +947, GPU +190, now: CPU 1137, GPU 257 (MiB)
Erron lightglue building
Error in SuperPoint building
十分感谢，请问这是什么意思

xukuanHIT, 2024-11-02:
可以 给出完整的log吗?

DeepestLover3685, 2024-11-02:
> 可以 给出完整的log吗？
[roslaunch-llk-27963.log](https://github.com/user-attachments/files/17606194/roslaunch-llk-27963.log)

十分感谢，麻烦您了

DeepestLover3685, 2024-11-02:
> 可以 给出完整的log吗?

[roslaunch-llk-27963.log](https://github.com/user-attachments/files/17606228/roslaunch-llk-27963.log)
十分感谢！

xukuanHIT, 2024-11-02:
这些是ros的log，屏幕种应该还有tensorRT相关的输出，也贴出来吧

DeepestLover3685, 2024-11-02:
(airgs) llk@llk:~/catkin_ws/src/air_slam/launch/visual_odometry$ roslaunch air_slam vo_euroc.launch
... logging to /home/llk/.ros/log/05840c56-98e9-11ef-80cc-e18d85137846/roslaunch-llk-27963.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://llk:42223/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.17.0
 * /visual_odometry/camera_config_path: /home/llk/catkin_...
 * /visual_odometry/config_path: /home/llk/catkin_...
 * /visual_odometry/dataroot: /home/llk/catkin_...
 * /visual_odometry/model_dir: /home/llk/catkin_...
 * /visual_odometry/saving_dir: /home/llk/catkin_...

NODES
  /
    rviz (rviz/rviz)
    visual_odometry (air_slam/visual_odometry)

ROS_MASTER_URI=http://localhost:11311

process[visual_odometry-1]: started with pid [27977]
process[rviz-2]: started with pid [27978]
config_file = /home/llk/catkin_ws/src/air_slam/configs/visual_odometry/vo_euroc.yaml
config done
[ INFO] [1730535477.780448188]: rviz version 1.14.25
[ INFO] [1730535477.780506598]: compiled against Qt version 5.12.8
[ INFO] [1730535477.780520701]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1730535477.788596294]: Forcing OpenGl version 0.
[ INFO] [1730535478.400025470]: Stereo is NOT SUPPORTED
[ INFO] [1730535478.400137640]: OpenGL device: Mesa Intel(R) UHD Graphics (CML GT2)
[ INFO] [1730535478.400176229]: OpenGl version: 4.6 (GLSL 4.6) limited to GLSL 1.4 on Mesa system.
[11/02/2024-16:17:59] [I] [TRT] [MemUsageChange] Init CUDA: CPU +14, GPU +0, now: CPU 35, GPU 67 (MiB)
[11/02/2024-16:18:01] [I] [TRT] [MemUsageChange] Init builder kernel library: CPU +947, GPU +190, now: CPU 1137, GPU 257 (MiB)
Erron lightglue building
Error in SuperPoint building
[visual_odometry-1] process has finished cleanly
log file: /home/llk/.ros/log/05840c56-98e9-11ef-80cc-e18d85137846/visual_odometry-1*.log
^C[rviz-2] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

xukuanHIT, 2024-11-02:
从输出没能看出来什么问题，你用docker能跑通吗？

DeepestLover3685, 2024-11-02:
> 从输出没能看出来什么问题，你用docker能跑通吗？

我还没有用过docker，现在就去试一下看看。我还需要更改TensorRT版本到8.5么？谢谢

xukuanHIT, 2024-11-02:
不需要的，AirVO用的是8.5，AirSLAM用8.6。另外，docker里面tensorRT已经配好8.6了

xukuanHIT, 2024-11-02:
你是什么显卡？

DeepestLover3685, 2024-11-02:
NVIDIA GeForce GTX 1650 Ti

xukuanHIT, 2024-11-02:
那可能是显卡的问题，我们之前也有运到过TensorRT8.6在1xxx系列的显卡上不太兼容的问题。你可以先试试docker。我们目前也在开发一个纯点特征的系统，会降低对TensorRT版本的要求，以能够在更多GPU上运行

DeepestLover3685, 2024-11-02:
> 那可能是显卡的问题，我们之前也有运到过TensorRT8.6在1xxx系列的显卡上不太兼容的问题。你可以先试试docker。我们目前也在开发一个纯点特征的系统，会降低对TensorRT版本的要求，以能够在更多GPU上运行

好的，辛苦了，十分感谢解答，加油！

mike130201, 2024-11-13:
> @xukuanHIT 您好！我在roslaunch air_slam vo_euroc.launch时遇到了错误。 首先是rviz没有显示 [...]

I encountered the same problem, and found a solution that worked well in Docker. Here’s what helped:

1.     Use Docker: Running the simulation in Docker resolved many environment-related issues.

2.     Ensure Proper Permissions: Make sure Docker has the necessary permissions to access the files, especially for directories like dataroot.

3.    Check Paths: Double-check that dataroot and other paths in the configuration files

after that I tried again

$ roslaunch air_slam vo_euroc.launch

DeepestLover3685, 2024-11-13:
> > 您好！我在roslaunch air_slam vo_euroc.launch时遇到了错误。首先是rviz没有显示 ![image](https://private-user-images.githubusercontent.com/143925927/382456680-5c089b5e-f9a8-409d-bcc5-c21728f19e27.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE0NDk2ODAsIm5iZiI6MTczMTQ0OTM4MCwicGF0aCI6Ii8xNDM5MjU5MjcvMzgyNDU2NjgwLTVjMDg5YjVlLWY5YTgtNDA5ZC1iY2M1LWMyMTcyOGYxOWUyNy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTEyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTExMlQyMjA5NDBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01NGFiMmRlNzFhODkxNTA3N2JmMjZiOTUwZDNhNDEzMDViODQzY2Y4YzFiOWZhZTQ4N2QzZTU4OTMyNzIyYjIyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.zwE6QK-C5dhRjtIFSoCl-DPSJrxkt47_cvYIy9X5Xn0) 终端有这样的报错 Erron lightglue building Error in SuperPoint building
> > 这是我vo_euroc.launch的配置以及dataroot路径![image](https://private-user-images.githubusercontent.com/143925927/382456714-255f797f-00c5-40b3-af69-4802feea95b7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE0NDk2ODAsIm5iZiI6MTczMTQ0OTM4MCwicGF0aCI6Ii8xNDM5MjU5MjcvMzgyNDU2NzE0LTI1NWY3OTdmLTAwYzUtNDBiMy1hZjY5LTQ4MDJmZWVhOTViNy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTEyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTExMlQyMjA5NDBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mOGQwM2VhZDQwNDFiOTk5ODUzNmQ1OGZmNDllOWM1ZThhMDRkN2FjYjZhNDc3ODYwZDhjMzVkNTViZDI3ZmZjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.awSlewPNaP6l8UhYRofIfDaqzVb1nZtuxJoSMwYf0yo)
> 
> 我遇到了同样的问题，并找到了一个在 Docker 中运行良好的解决方案。以下是有帮助的：
> 
> 1. ```
>    Use Docker: Running the simulation in Docker resolved many environment-related issues.
>    ```
> 2. ```
>    Ensure Proper Permissions: Make sure Docker has the necessary permissions to access the files, especially for directories like dataroot.
>    ```
> 3. 检查路径：仔细检查配置文件中的 dataroot 和其他路径
> 
> 之后我又试了一次
> 
> $ roslaunch air_slam vo_euroc.launch

好的，我更换了40系列也是出现了一些问题，现在打算使用docker

DeepestLover3685, 2024-11-15:
> > 您好！我在roslaunch air_slam vo_euroc.launch时遇到了错误。首先是rviz没有显示 ![图像](https://private-user-images.githubusercontent.com/143925927/382456680-5c089b5e-f9a8-409d-bcc5-c21728f19e27.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE0NDk2ODAsIm5iZiI6MTczMTQ0OTM4MCwicGF0aCI6Ii8xNDM5MjU5MjcvMzgyNDU2NjgwLTVjMDg5YjVlLWY5YTgtNDA5ZC1iY2M1LWMyMTcyOGYxOWUyNy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTEyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTExMlQyMjA5NDBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01NGFiMmRlNzFhODkxNTA3N2JmMjZiOTUwZDNhNDEzMDViODQzY2Y4YzFiOWZhZTQ4N2QzZTU4OTMyNzIyYjIyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.zwE6QK-C5dhRjtIFSoCl-DPSJrxkt47_cvYIy9X5Xn0) 终端有这样的报错 Erron lightglue building Error in SuperPoint building
> > 这是我vo_euroc.launch的配置以及dataroot路径![图像](https://private-user-images.githubusercontent.com/143925927/382456714-255f797f-00c5-40b3-af69-4802feea95b7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE0NDk2ODAsIm5iZiI6MTczMTQ0OTM4MCwicGF0aCI6Ii8xNDM5MjU5MjcvMzgyNDU2NzE0LTI1NWY3OTdmLTAwYzUtNDBiMy1hZjY5LTQ4MDJmZWVhOTViNy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTEyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTExMlQyMjA5NDBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mOGQwM2VhZDQwNDFiOTk5ODUzNmQ1OGZmNDllOWM1ZThhMDRkN2FjYjZhNDc3ODYwZDhjMzVkNTViZDI3ZmZjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.awSlewPNaP6l8UhYRofIfDaqzVb1nZtuxJoSMwYf0yo)
> 
> 我遇到了同样的问题，并找到了一个在 Docker 中运行良好的解决方案。以下是有帮助的：
> 
> 1. ```
>    Use Docker: Running the simulation in Docker resolved many environment-related issues.
>    ```
> 2. ```
>    Ensure Proper Permissions: Make sure Docker has the necessary permissions to access the files, especially for directories like dataroot.
>    ```
> 3. 检查路径：仔细检查配置文件中的 dataroot 和其他路径
> 
> 之后我又试了一次
> 
> $ roslaunch air_slam vo_euroc.launch

您好，可以看一下您的路径配置么。我使用MN_015_GV_01数据集可以跑vo_euroc.launch，但是没有现象

mike130201, 2024-11-20:
> 您好，可以看一下您的路径配置么。我使用MN_015_GV_01数据集可以跑vo_euroc.launch，但是没有现象

I don't know If I translate correctly your message, but here is my dataroot:
/workspace/catkin_ws_1/src/V1_02_medium/mav0/

Alranen, 2025-02-18:
您好，我想将之间的双目数据集（没有IMU数据）转换为Euroc的格式运行，euroc.yaml中的：
# Calibration
distortion_type: 1  # 0 for undistorted inputs, 1 for radial-tangential: [k1, k2, p1, p2, k3], 2 for equidistant/fisheye:  [k1, k2, k3, k4, 0].
cam0:
  intrinsics: [458.654, 457.296, 367.215, 248.375] # fx, fy, cx, cy
  distortion_coeffs: [-0.28340811, 0.07395907, 0.00019359, 1.76187114e-05, 0]
  T_type: 0           # 0 for Euroc format, the following T is Tbc. 1 for Kalibr format, the following T is Tcb
  T: 
  - [0.0148655429818, -0.999880929698, 0.00414029679422, -0.0216401454975]
  - [0.999557249008, 0.0149672133247, 0.025715529948, -0.064676986768]
  - [-0.0257744366974, 0.00375618835797, 0.999660727178, 0.00981073058949]
  - [0.0, 0.0, 0.0, 1.0]
cam1:
  intrinsics: [457.587, 456.134, 379.999, 255.238] 
  distortion_coeffs: [-0.28368365,  0.07451284, -0.00010473, -3.55590700e-05, 0]
  T_type: 0           
  T: 
  - [0.0125552670891, -0.999755099723, 0.0182237714554, -0.0198435579556]
  - [0.999598781151, 0.0130119051815, 0.0251588363115, 0.0453689425024]
  - [-0.0253898008918, 0.0179005838253, 0.999517347078, 0.00786212447038]
  - [0.0, 0.0, 0.0, 1.0]
“cam0: T”和“cam1: T” 应该怎么设置，ORB-SLAM3中只有一个Stereo.T_c1_c2

xukuanHIT, 2025-02-19:
@Alranen 可以把左目设成单位阵