sair-lab/AirSLAM #163: Generate SuperGlue.engine problem

cpymaple, 2024-11-04:
Hi, I'm encountering an issue where, despite keeping all dependencies, code, dataset, and hardware unchanged, the size of the generated **SuperGlue.engine** file varies each time. Occasionally, the generated .engine file works as expected, performing matching correctly. However, sometimes it fails to produce any matches.

1. Have you experienced a similar issue with inconsistent .engine file sizes and functionality?
2. Are there specific parameters that need to be adjusted to ensure stability in each generated .engine file?
3. Additionally, I’ve written a ROS interface for AirSLAM (similar to AirVO) and play the euroc rosbag to directly generate .engine file. Is this a correct method, or could it be contributing to the inconsistencies?

Any insights or suggestions would be greatly appreciated. Thank you!

xukuanHIT, 2024-11-08:
@cpymaple Hi, sorry for the delayed response. These two lines ([L1](https://github.com/sair-lab/AirSLAM/blob/2a23e669f20da45824df486c6b326d84f4fbfb67/src/light_glue.cpp#L60) and [L2](https://github.com/sair-lab/AirSLAM/blob/2a23e669f20da45824df486c6b326d84f4fbfb67/src/light_glue.cpp#L64)) define the maximum size for the input image. If the image is too large, the generated .engine file may encounter issues. Therefore, for high-resolution images, these two parameters need to be increased before generating the .engine file.

cpymaple, 2024-11-12:
> @cpymaple Hi, sorry for the delayed response. These two lines ([L1](https://github.com/sair-lab/AirSLAM/blob/2a23e669f20da45824df486c6b326d84f4fbfb67/src/light_glue.cpp#L60) and [...]

Hi, Thanks for your suggestions. For me, it seems that I used the wrong dependency for generating the .onnx file (cuda 11.8). Now I use the TensorRT 8.4.1.5 and cuda 11.6 to generate .onnx and .engine file, it works for me. Theoretically, it should be also work in TensorRT 8.6 and cuda 12.1.

closed 2024-11-12 by cpymaple