sair-lab/AirSLAM #166: Use ORB-SLAM 3 to test and run the OIVIO dataset

nishibenbi, 2024-11-11:
你好由于硬件问题我目前想先用ORB-SLAM３运行下OIVIO，后续再在AirSLAM上运行进行对比，但是不知是配置文件问题或是数据集路径问题一直不能运行成功，请问这方面应该如何设置，您那边是否有这方面的程序文件提供
[image]

xukuanHIT, 2024-11-18:
在加载字典吗？

nishibenbi, 2024-11-18:
似乎是图像没有加载成功，一直卡在这个界面没有报错也没有可视化界面输出，这里是不是跟我用的TUM-VI的启动文件有关呢，请问您当时用ORB-SLAM3运行OIVIO是用自己调试的文件吗

xukuanHIT, 2024-11-18:
我们在euroc上改的，不过用哪个都可以的。你这个问题得认真debug一下卡在了哪里

sukaixiong, 2024-11-20:
> 你好由于硬件问题我目前想先用ORB-SLAM３运行下OIVIO，后续再在AirSLAM上运行进行对比，但是不知是配置文件问题或是数据集路径问题一直不能运行成功，请问这方面应该如何设置，您那边是否有这方面的程序文件提供 <img alt="图片20241111203707" width="490" src="https://private-user-images.githubusercontent.com/106231861/384925309-3d63d9d8-81f6-4cf3-bc53-b24df5fac961.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzIwNjQ5NjQsIm5iZiI6MTczMjA2NDY2NCwicGF0aCI6Ii8xMDYyMzE4NjEvMzg0OTI1MzA5LTNkNjNkOWQ4LTgxZjYtNGNmMy1iYzUzLWIyNGRmNWZhYzk2MS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTIwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEyMFQwMTA0MjRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00ZTNhNGI1MDNjZWQwYTdkMTg0YmE3YWUxZjFkYTVhMzY5YTZlZWI3NTNlN2YzNDNkZjc0ZTI5NWRlOTA2NmExJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.4zSGqA6Uyp8r0Ndouf5

> 你好由于硬件问题我目前想先用ORB-SLAM３运行下OIVIO，后续再在AirSLAM上运行进行对比，但是不知是配置文件问题或是数据集路径问题一直不能运行成功，请问这方面应该如何设置，您那边是否有这方面的程序文件提供 <img alt="图片20241111203707" width="490" src="https://private-user-images.githubusercontent.com/106231861/384925309-3d63d9d8-81f6-4cf3-bc53-b24df5fac961.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzIwNjQ5NjQsIm5iZiI6MTczMjA2NDY2NCwicGF0aCI6Ii8xMDYyMzE4NjEvMzg0OTI1MzA5LTNkNjNkOWQ4LTgxZjYtNGNmMy1iYzUzLWIyNGRmNWZhYzk2MS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTIwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEyMFQwMTA0MjRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00ZTNhNGI1MDNjZWQwYTdkMTg0YmE3YWUxZjFkYTVhMzY5YTZlZWI3NTNlN2YzNDNkZjc0ZTI5NWRlOTA2NmExJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.4zSGqA6Uyp8r0Ndouf5KOROtuVW1ipfOEtkf97n3U6U">

你好，OIVIO数据集的官方链接没了，你可以给我发一份吗，2656419064@qq.com.十分感谢

nishibenbi, 2024-11-20:
作者给出的部分OIVIO数据集链接
https://entuedu-my.sharepoint.com/personal/kuan_xu_staff_main_ntu_edu_sg/_layouts/15/onedrive.aspx?id=%2Fpersonal%2Fkuan%5Fxu%5Fstaff%5Fmain%5Fntu%5Fedu%5Fsg%2FDocuments%2Fdatasets%2Foivio&amp;ga=1

朱澳苏
***@***.***

sukaixiong, 2024-11-30:
[image]
还是这个无法访问

sukaixiong, 2024-12-05:
有个问题，oivio的数据集的真值如何转换成tum格式呢？

sukaixiong, 2024-12-05:
> 我们在euroc上改的，不过用哪个都可以的。你这个问题得认真debug一下卡在了哪里
有个问题，oivio数据集的真值每行只有四列，真是如何转成tum格式的呢

xukuanHIT, 2024-12-17:
我们把旋转补成了单位阵，然后用evo只测了平移误差

laplace-pym, 2025-01-21:
> > 你好由于硬件问题我目前想先用ORB-SLAM３运行下OIVIO，后续再在AirSLAM上运行进行对比，但是不知是配置文件问题或是数据集路径问题一直不能运行成功，请问这方面应该如何设置，您那边是否有这方面的程序文件提供 <img alt="图片20241111203707" width="490" src="https://private-user-images.githubusercontent.com/106231861/384925309-3d63d9d8-81f6-4cf3-bc53-b24df5fac961.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzIwNjQ5NjQsIm5iZiI6MTczMjA2NDY2NCwicGF0aCI6Ii8xMDYyMzE4NjEvMzg0OTI1MzA5LTNkNjNkOWQ4LTgxZjYtNGNmMy1iYzUzLWIyNGRmNWZhYzk2MS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTIwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEyMFQwMTA0MjRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00ZTNhNGI1MDNjZWQwYTdkMTg0YmE3YWUxZjFkYTVhMzY5YTZlZWI3NTNlN2YzNDNkZjc0ZTI5NWRlOTA2NmExJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.4zSGqA6Uyp8r0Ndouf5
> 
> > 你好由于硬件问题我目前想先用ORB-SLAM３运行下OIVIO，后续再在AirSLAM上运行进行对比，但是不知是配置文件问题或是数据集路径问题一直不能运行成功，请问这方面应该如何设置，您那边是否有这方面的程序文件提供 <img alt="图片20241111203707" width="490" src="https://private-user-images.githubusercontent.com/106231861/384925309-3d63d9d8-81f6-4cf3-bc53-b24df5fac961.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzIwNjQ5NjQsIm5iZiI6MTczMjA2NDY2NCwicGF0aCI6Ii8xMDYyMzE4NjEvMzg0OTI1MzA5LTNkNjNkOWQ4LTgxZjYtNGNmMy1iYzUzLWIyNGRmNWZhYzk2MS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTIwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEyMFQwMTA0MjRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00ZTNhNGI1MDNjZWQwYTdkMTg0YmE3YWUxZjFkYTVhMzY5YTZlZWI3NTNlN2YzNDNkZjc0ZTI5NWRlOTA2NmExJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.4zSGqA6Uyp8r0Ndouf5KOROtuVW1ipfOEtkf97n3U6U">
> 
> 你好，OIVIO数据集的官方链接没了，你可以给我发一份吗，[2656419064@qq.com](mailto:2656419064@qq.com).十分感谢

您好，请问您现在有OIVIO的数据集吗，能否发我一份，这是我的邮箱1721179144@qq.com

flowerlegion, 2025-02-05:
> 你好，OIVIO数据集的官方链接没了，你可以给我发一份吗，[2656419064@qq.com](mailto:2656419064@qq.com).十分感谢

你好，请问您现在还有OVIO的数据集吗，可以给我发一份嘛，1076497434@qq.com,十分感谢

xukuanHIT, 2025-02-16:
@laplace-pym @flowerlegion 可以试试这个[链接](https://entuedu-my.sharepoint.com/:f:/g/personal/kuan_xu_staff_main_ntu_edu_sg/EtabKNOPBPBMmjJXbN4A8wMB0MTZmeZlnfAi7bskgfAr7A?e=Y9b4VL)下载OIVIO数据集。