sair-lab/AirSLAM #173: Intelsense D455 error

jsreveal, 2024-11-27:
Hi there. You have mentioned in another issue that you tested with Realsense D455 without noticing major issues. I am testing now with RealSense D455, but I am experiencing errors in specially in the angle of turning.

Which topics/streams did you use? I am using the two stereo infra1 and infra2, with the following calibration config. It is able to detect many stereo points, so I assume config is ok:
[image]

basically I recorded supermarket corridors that are of the same size, and perfectly aligned in parallel to each other, but the odometry result is the following:

[image]

Thank you!

xukuanHIT, 2024-11-28:
@jsreveal Hi, I think there is an error in either the image_width or the cx parameter you wrote. cx should roughly be half of the image_width.

jsreveal, 2024-11-28:
oh man! thanks a lot. I was using the intrinsics from the 640x480 by mistake. Now this looks like something!

closed 2024-11-29 by jsreveal