sair-lab/AirSLAM #183: catkin_make遇到了问题

medsadw, 2025-01-17:
您好，我在catkin_make的过程中遇到了问题。从报错信息我认为可能是TensorRT的问题，我的TensorRT为8.6.1.6，并在CmakeList.txt设置了TensorRT文件的路径。
`include_directories(/home/csw/downloads/TensorRT-8.6.1.6/include) 
link_directories(/home/csw/downloads/TensorRT-8.6.1.6/bin) `

但是对于3rdparty/tensorrtbuffer的编译依然遇到了下面的报错信息 
`[  1%] Building CXX object AirSLAM/3rdparty/tensorrtbuffer/CMakeFiles/tensorrtbuffer.dir/src/logger.cpp.o
[  3%] Building CXX object AirSLAM/3rdparty/tensorrtbuffer/CMakeFiles/tensorrtbuffer.dir/src/sample_options.cpp.o
[  5%] Building CXX object AirSLAM/3rdparty/tensorrtbuffer/CMakeFiles/tensorrtbuffer.dir/src/sample_utils.cpp.o
[ 19%] Built target DBoW2
In file included from /home/csw/Air_SLAM_ws/src/AirSLAM/3rdparty/tensorrtbuffer/src/logger.cpp:19:
/home/csw/Air_SLAM_ws/src/AirSLAM/3rdparty/tensorrtbuffer/include/error_recorder.h: In member function ‘virtual void TensorRTErrorRecorder::clear()’:
/home/csw/Air_SLAM_ws/src/AirSLAM/3rdparty/tensorrtbuffer/include/error_recorder.h:78:13: error: ‘tensorrt_log’ has not been declared
   78 |             tensorrt_log::gLogFatal << "Internal Error: " << e.what() << std::endl;
      |             ^~~~~~~~~~~~
/home/csw/Air_SLAM_ws/src/AirSLAM/3rdparty/tensorrtbuffer/include/error_recorder.h: In member function ‘virtual bool TensorRTErrorRecorder::reportError(nvinfer1::ErrorCode, nvinfer1::IErrorRecorder::ErrorDesc)’:
/home/csw/Air_SLAM_ws/src/AirSLAM/3rdparty/tensorrtbuffer/include/error_recorder.h:93:13: error: ‘tensorrt_log’ has not been declared
   93 |             tensorrt_log::gLogError << "Error[" << static_cast<int32_t>(val) << "]: " << desc << std::endl;
      |             ^~~~~~~~~~~~
/home/csw/Air_SLAM_ws/src/AirSLAM/3rdparty/tensorrtbuffer/include/error_recorder.h:98:13: error: ‘tensorrt_log’ has not been declared
   98 |             tensorrt_log::gLogFatal << "Internal Error: " << e.what() << std::endl;
      |             ^~~~~~~~~~~~
/home/csw/Air_SLAM_ws/src/AirSLAM/3rdparty/tensorrtbuffer/src/logger.cpp: At global scope:
/home/csw/Air_SLAM_ws/src/AirSLAM/3rdparty/tensorrtbuffer/src/logger.cpp:22:23: error: conflicting declaration ‘TensorRTErrorRecorder gRecorder’
   22 | TensorRTErrorRecorder gRecorder;
      |                       ^~~~~~~~~
In file included from /home/csw/Air_SLAM_ws/src/AirSLAM/3rdparty/tensorrtbuffer/src/logger.cpp:18:
/home/csw/downloads/TensorRT-8.6.1.6/samples/common/logger.h:24:28: note: previous declaration as ‘SampleErrorRecorder gRecorder’
   24 | extern SampleErrorRecorder gRecorder;
      |                            ^~~~~~~~~
/home/csw/Air_SLAM_ws/src/AirSLAM/3rdparty/tensorrtbuffer/src/logger.cpp:25:5: error: ‘Logger’ does not name a type; did you mean ‘getLogger’?
   25 |     Logger gLogger{Logger::Severity::kINFO};
      |     ^~~~~~
      |     getLogger
/home/csw/Air_SLAM_ws/src/AirSLAM/3rdparty/tensorrtbuffer/src/logger.cpp:26:5: error: ‘LogStreamConsumer’ does not name a type
   26 |     LogStreamConsumer gLogVerbose{LOG_VERBOSE(gLogger)};
      |     ^~~~~~~~~~~~~~~~~
/home/csw/Air_SLAM_ws/src/AirSLAM/3rdparty/tensorrtbuffer/src/logger.cpp:27:5: error: ‘LogStreamConsumer’ does not name a type
   27 |     LogStreamConsumer gLogInfo{LOG_INFO(gLogger)};
      |     ^~~~~~~~~~~~~~~~~
/home/csw/Air_SLAM_ws/src/AirSLAM/3rdparty/tensorrtbuffer/src/logger.cpp:28:5: error: ‘LogStreamConsumer’ does not name a type
   28 |     LogStreamConsumer gLogWarning{LOG_WARN(gLogger)};
      |     ^~~~~~~~~~~~~~~~~
/home/csw/Air_SLAM_ws/src/AirSLAM/3rdparty/tensorrtbuffer/src/logger.cpp:29:5: error: ‘LogStreamConsumer’ does not name a type
   29 |     LogStreamConsumer gLogError{LOG_ERROR(gLogger)};
      |     ^~~~~~~~~~~~~~~~~
/home/csw/Air_SLAM_ws/src/AirSLAM/3rdparty/tensorrtbuffer/src/logger.cpp:30:5: error: ‘LogStreamConsumer’ does not name a type
   30 |     LogStreamConsumer gLogFatal{LOG_FATAL(gLogger)};
      |     ^~~~~~~~~~~~~~~~~
/home/csw/Air_SLAM_ws/src/AirSLAM/3rdparty/tensorrtbuffer/src/logger.cpp:32:32: error: variable or field ‘setReportableSeverity’ declared void
   32 |     void setReportableSeverity(Logger::Severity severity)
      |                                ^~~~~~
/home/csw/Air_SLAM_ws/src/AirSLAM/3rdparty/tensorrtbuffer/src/logger.cpp:32:32: error: ‘Logger’ has not been declared
make[2]: *** [AirSLAM/3rdparty/tensorrtbuffer/CMakeFiles/tensorrtbuffer.dir/build.make:76：AirSLAM/3rdparty/tensorrtbuffer/CMakeFiles/tensorrtbuffer.dir/src/logger.cpp.o] 错误 1
make[2]: *** 正在等待未完成的任务....
In file included from /home/csw/Air_SLAM_ws/src/AirSLAM/3rdparty/tensorrtbuffer/include/common.h:62,
                 from /home/csw/Air_SLAM_ws/src/AirSLAM/3rdparty/tensorrtbuffer/include/sample_utils.h:35,
                 from /home/csw/Air_SLAM_ws/src/AirSLAM/3rdparty/tensorrtbuffer/src/sample_options.cpp:31:
/home/csw/Air_SLAM_ws/src/AirSLAM/3rdparty/tensorrtbuffer/include/safe_common.h: In member function ‘void tensorrt_buffer::TrtCudaGraphSafe::endCaptureOnError(CUstream_st*&)’:
/home/csw/Air_SLAM_ws/src/AirSLAM/3rdparty/tensorrtbuffer/include/safe_common.h:182:13: error: ‘tensorrt_log’ has not been declared
  182 |             tensorrt_log::gLogError << "The CUDA graph capture on the stream has failed." << std::endl;
      |             ^~~~~~~~~~~~
/home/csw/Air_SLAM_ws/src/AirSLAM/3rdparty/tensorrtbuffer/include/safe_common.h: In function ‘void tensorrt_buffer::safeLoadLibrary(const string&)’:
/home/csw/Air_SLAM_ws/src/AirSLAM/3rdparty/tensorrtbuffer/include/safe_common.h:203:13: error: ‘tensorrt_log’ has not been declared
  203 |             tensorrt_log::gLogError << "Could not load plugin library: " << path << ", due to: " << dlerror() << std::endl;
      |             ^~~~~~~~~~~~
In file included from /home/csw/Air_SLAM_ws/src/AirSLAM/3rdparty/tensorrtbuffer/include/sample_utils.h:35,
                 from /home/csw/Air_SLAM_ws/src/AirSLAM/3rdparty/tensorrtbuffer/src/sample_options.cpp:31:
/home/csw/Air_SLAM_ws/src/AirSLAM/3rdparty/tensorrtbuffer/include/common.h: In function ‘int32_t tensorrt_buffer::calculateSoftmax(float*, int32_t)’:
/home/csw/Air_SLAM_ws/src/AirSLAM/3rdparty/tensorrtbuffer/include/common.h:91:13: error: ‘tensorrt_log’ has not been declared
   91 |             tensorrt_log::gLogError << "Assertion failure: " << #condition << std::endl;  \
[... 535 lines cut ...]
/home/csw/Air_SLAM_ws/src/AirSLAM/3rdparty/tensorrtbuffer/src/sample_options.cpp:657:59: error: ‘BuildOptions’ has not been declared
  657 |         std::ostream& printPreviewFlags(std::ostream& os, BuildOptions const& options)
      |                                                           ^~~~~~~~~~~~
/home/csw/Air_SLAM_ws/src/AirSLAM/3rdparty/tensorrtbuffer/src/sample_options.cpp: In function ‘std::ostream& tensorrt_log::{anonymous}::printPreviewFlags(std::ostream&, const int&)’:
/home/csw/Air_SLAM_ws/src/AirSLAM/3rdparty/tensorrtbuffer/src/sample_options.cpp:659:25: error: request for member ‘previewFeatures’ in ‘options’, which is of non-class type ‘const int’
  659 |             if (options.previewFeatures.empty())
      |                         ^~~~~~~~~~~~~~~
/home/csw/Air_SLAM_ws/src/AirSLAM/3rdparty/tensorrtbuffer/src/sample_options.cpp: In lambda function:
/home/csw/Air_SLAM_ws/src/AirSLAM/3rdparty/tensorrtbuffer/src/sample_options.cpp:667:29: error: request for member ‘previewFeatures’ in ‘options’, which is of non-class type ‘const int’
  667 |                 if (options.previewFeatures.find(featVal) != options.previewFeatures.end())
      |                             ^~~~~~~~~~~~~~~
/home/csw/Air_SLAM_ws/src/AirSLAM/3rdparty/tensorrtbuffer/src/sample_options.cpp:667:70: error: request for member ‘previewFeatures’ in ‘options’, which is of non-class type ‘const int’
  667 |                 if (options.previewFeatures.find(featVal) != options.previewFeatures.end())
      |                                                                      ^~~~~~~~~~~~~~~
/home/csw/Air_SLAM_ws/src/AirSLAM/3rdparty/tensorrtbuffer/src/sample_options.cpp:669:68: error: request for member ‘previewFeatures’ in ‘options’, which is of non-class type ‘const int’
  669 |                     os << previewFeatureToString(feat) << (options.previewFeatures.at(featVal) ? " [ON], " : " [OFF], ");
      |                                                                    ^~~~~~~~~~~~~~~
/home/csw/Air_SLAM_ws/src/AirSLAM/3rdparty/tensorrtbuffer/src/sample_options.cpp: At global scope:
/home/csw/Air_SLAM_ws/src/AirSLAM/3rdparty/tensorrtbuffer/src/sample_options.cpp:682:5: error: ‘Arguments’ does not name a type

make[2]: *** [AirSLAM/3rdparty/tensorrtbuffer/CMakeFiles/tensorrtbuffer.dir/build.make:90：AirSLAM/3rdparty/tensorrtbuffer/CMakeFiles/tensorrtbuffer.dir/src/sample_options.cpp.o] 错误 1
make[1]: *** [CMakeFiles/Makefile2:1921：AirSLAM/3rdparty/tensorrtbuffer/CMakeFiles/tensorrtbuffer.dir/all] 错误 2
make: *** [Makefile:146：all] 错误 2
Invoking "make -j24 -l24" failed
`

xukuanHIT, 2025-01-18:
看着像是环境没有配置好，用docker能跑通吗？

medsadw, 2025-01-20:
> 看着像是环境没有配置好，用docker能跑通吗？

谢谢您的回答！我是在主机环境上跑的，我尝试一下用docker来跑

Mcaft, 2026-10-12:
能是能跑，但是对TRT版本要求有点严格。应该是Jetpack6.0适配的。而且blog写的那个环境，ubuntu2204+noetic，ros1装ros2的依赖，也太怪了